bloombloombloom/Bloom #8: "JTAGM failed to initialise" - Atmel-ICE, UPDI, ATmega4809

timwalls, 2021-10-30:
Hi!

Super excited about the possibility of Bloom - support for '4809 is universally terrible, so anything that improves that will be awesome - but running into some initial problems, maybe you can help.

Using an Atmel-ICE, with a Microchip atmega4809Xplained evaluation board; debug interface is UPDI.  Running bloom gives me the following error:

```
2021-10-30 15:46:59 UTC [MT] [1]: [INFO] Selected environment: default
2021-10-30 15:46:59 UTC [TC] [5]: [INFO] Starting TargetController
2021-10-30 15:46:59 UTC [TC] [6]: [INFO] Connecting to debug tool
2021-10-30 15:46:59 UTC [TC] [8]: [INFO] Debug tool connected
2021-10-30 15:46:59 UTC [TC] [9]: [INFO] Debug tool name: Atmel-ICE
2021-10-30 15:46:59 UTC [TC] [10]: [INFO] Debug tool serial: J42700029646
2021-10-30 15:46:59 UTC [TC] [14]: [INFO] Activating target
2021-10-30 15:46:59 UTC [TC] [15]: [ERROR] The TargetController encountered a fatal error. See below for errors:
2021-10-30 15:46:59 UTC [TC] [16]: [ERROR] Failed to set parameter on device! - Failure reason: JTAGM failed to initialise
2021-10-30 15:46:59 UTC [TC] [17]: [INFO] Shutting down TargetController
2021-10-30 15:46:59 UTC [TC] [18]: [INFO] Deactivating target
2021-10-30 15:46:59 UTC [TC] [19]: [INFO] Closing debug tool
2021-10-30 15:46:59 UTC [MT] [21]: [ERROR] TargetController failed to startup.
2021-10-30 15:46:59 UTC [MT] [22]: [INFO] Shutting down Bloom
2021-10-30 15:46:59 UTC [SH] [27]: [INFO] Shutting down SignalHandler
```

Version is `Bloom v0.4.1`.  My config file looks like:

```
{
  "environments": {
    "default": {
      "debugTool": {
        "name": "atmel-ice",
        "releasePostDebugSession": true
      },

      "target": {
        "name": "atmega4809",
        "physicalInterface": "updi"
      },

      "debugServer": {
        "name": "avr-gdb-rsp",
        "ipAddress": "0.0.0.0",
        "port": "1442"
      }
    }
  },

  "insight": {
    "enabled": true
  }
}
```

I can confirm the hardware & debugging all works with the dreaded Microchip Studio on a Windows host, so not a hardware issue.

Any thoughts, or anything I can do to help debug?

Cheers,
Tim

navnavnav, 2021-10-30:
Hi Tim,

Ah, I think I see the problem here. Bloom was failing to resolve the correct config variant for the debug session. This meant that it did not configure the UPDI clock speed, before attempting to set parameters on the debug tool.

I've applied the fix for this and rebuilt Bloom. I don't have an ATmega4809 to test with, so I've attached a link to the new build to this comment. Please remove your old build (v0.4.1) and install this build (v0.4.2).

New build (v0.4.2):
https://bloom-public-bucket.s3.eu-west-1.amazonaws.com/misc/Bloom-0.4.2-Linux.deb

You can remove your old build by running
```
sup apt-get remove bloom
```
Then install the new build via
```
sudo apt install /path/to/new-build.deb
```
(replace the path in the above command)

Then try to run Bloom again, from within your project directory. Please post an update on this issue so I can decide whether to create an official release with this fix.

Thanks,
Nav

timwalls, 2021-10-31:
I am genuinely super excited to let you know - yes, this seems to work now:

```
2021-10-31 09:53:22 UTC [MT] [1]: [INFO] Selected environment: default
2021-10-31 09:53:22 UTC [TC] [5]: [INFO] Starting TargetController
2021-10-31 09:53:22 UTC [TC] [6]: [INFO] Connecting to debug tool
2021-10-31 09:53:22 UTC [TC] [8]: [INFO] Debug tool connected
2021-10-31 09:53:22 UTC [TC] [9]: [INFO] Debug tool name: Atmel-ICE
2021-10-31 09:53:22 UTC [TC] [10]: [INFO] Debug tool serial: J42700029646
2021-10-31 09:53:22 UTC [TC] [14]: [INFO] Activating target
2021-10-31 09:53:23 UTC [TC] [29]: [INFO] Target activated
2021-10-31 09:53:23 UTC [TC] [30]: [INFO] AVR8 target promoted to megaAVR target
2021-10-31 09:53:24 UTC [TC] [45]: [INFO] Target ID: 0x1e9651
2021-10-31 09:53:24 UTC [TC] [46]: [INFO] Target name: ATmega4809
2021-10-31 09:53:24 UTC [MT] [49]: [INFO] Selected DebugServer: AVR GDB Remote Serial Protocol Debug Server
2021-10-31 09:53:24 UTC [DS] [50]: [INFO] Starting DebugServer
2021-10-31 09:53:24 UTC [DS] [55]: [INFO] GDB RSP address: 0.0.0.0
2021-10-31 09:53:24 UTC [DS] [56]: [INFO] GDB RSP port: 1442
2021-10-31 09:53:24 UTC [DS] [58]: [INFO] DebugServer ready
2021-10-31 09:53:24 UTC [DS] [59]: [INFO] Waiting for GDB RSP connection
2021-10-31 09:53:24 UTC [DS] [61]: [INFO] Waiting for GDB RSP connection
2021-10-31 09:53:24 UTC [MT] [62]: [INFO] Starting Insight
2021-10-31 09:53:24 UTC [MT] [67]: [INFO] Insight ready
```

And a screenshot to prove it: https://imgur.com/a/QbXmaWm

I can't tell you how awesome this is, finding working debug tools for the '4809 is next to impossible, so this is going to make a huge difference.  Let me know if I can buy you a coffee somehow :-D.

navnavnav, 2021-10-31:
That's excellent news, thanks for confirming, Tim,

I've just published the new release as v0.4.2, with yourself being credited in the release notes for reporting the issue. The release is now live on the Bloom website.

Also, just FYI, I noticed that Bloom's TDF for the ATmega4809 was missing the PDIP package variant definition, so I've included that in the 0.4.2 release. But that was after building the release which was linked to my previous comment. So if you want to use the PDIP variant in the Insight window, you'll have to download the official version of the release from the website and then follow the same instructions as specified in my previous comment.

I'm going to close this issue now. Please let me know if you have any other issues with Bloom.

closed 2021-10-31 by navnavnav