bloombloombloom/Bloom #9: Distribute Bloom in RPM package

navnavnav, 2021-11-11:
Bloom is currently distributed via a Debian package. Users of non-debian based distributions are unable to install the prebuilt Bloom package. They are forced to build Bloom from source, which isn't ideal.

It would be nice if we could maintain an RPM package along with a Debian one.

lkoepsel, 2021-12-26:
Just started using Bloom with an Uno and a Snap. Just in case others have a similar issue. Regarding .deb package installation, I did the following for ArchLinux:

Using the instructions here, I converted the .deb to a tar.zst package
https://ostechnix.com/convert-deb-packages-arch-linux-packages/
```
sudo debtap -u
deptap Downloads/Bloom-0.4.2-Linux.deb
sudo pacman -U Downloads/bloom-0.4.2-1-x86_64.pkg.tar.zst
```
Installed with no issues.

pavelskipenes, 2022-03-30:
Could you consider shipping it as Flatpak in stead?

navnavnav, 2022-04-02:
Hi @pavelskipenes,

I've just shipped an RPM package and updated the website. I have tested the package on Fedora 35 and it works well. Please download the RPM via https://bloom.oscillate.io/download and let me know if it works on your distro. I will close this ticket once I've received confirmation from you.

Thanks

pavelskipenes, 2022-04-02:
@navnavnav I'm not on Fedora :/ I've "installed" it by unzipping the `.deb` package. Didn't manage it to run just yet but I assume it's because of configuration mismatch with my hardware. Anyway it's not related to this issue :)

navnavnav, 2022-04-02:
@pavelskipenes You don't have to be on Fedora. You can just download the RPM via the "Other" option and install it as you normally would:
[image]

But given that you've apparently installed it manually, I will now close this ticket.

closed 2022-04-02 by navnavnav

SamsiFPV, 2025-03-21:
Is there any plans to make 2.0 as a RPM as well? I can't even find the 1.0 RPM anymore.