bloombloombloom/Bloom #10: Implement device driver for Microchip Xplained Pro development boards

navnavnav, 2021-11-20:
The Microchip Xplained Pro development board comes with an embedded debugger, accessible over USB. The interface for the debugger appears to EDBG-based, so we should be able to use the `EdbgAvr8Interface` for interfacing with AVR8 targets.

mraardvark, 2021-11-21:
I think there are only a couple of AVR XPRO boards...
- tiny817
- mega324pb

The xplained mini series has:
- mega328p
- mega328pb
- mega168pb

there are a couple of xplained nanos
- tinyy416

Most interesting (for me) are the Curiosity Nanos
- tiny1607
- tiny1627
- mega4809
- avrda
- avrdb
- <insert new devices here>

navnavnav, 2021-11-22:
Thanks @mraardvark, I'm currently testing with the [ATMEGA4809-XPRO](https://uk.farnell.com/microchip/atmega4809-xpro/dev-board-8bit-atmega-avr-mcu/dp/2857514) board, which is marketed as an Xplained Pro board. It also uses the `0x2111` USB product ID.

Once I have the Xplained Pro working properly, I will consider adding support for others.

I have the board working with Bloom, but I spent most of the weekend trying to figure out the cause for an issue with reading target SRAM larger than 256 bytes. For some reason, the debug tool returns incorrect data when reading more than 256 bytes from SRAM, via a single read memory command:

```
0x21 0x00 0x20 0x00 0x00 0x00 0x00 0x01 0x01 0x00 0x00
```

The above command requests 257 bytes of SRAM, starting at address `0x00`. As the command does not exceed the HID report size of 512 bytes, the Xplained Pro debug tool responds with a single response, containing no error or any indication of an error. It responds with a response code of `0x84` (indicating a data response). The problem is that the data it returns is completely incorrect. For the mega4809, the address `0x3D` and `0x3E` should contain the stack pointer HIGH and LOW registers, respectively. But looking at the data in the response, no such values are present at those addresses.

Now consider the following:

```
0x21 0x00 0x20 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x00
```

The above command is very similar to the previous. The only difference being the size of memory being requested: 256 bytes, as opposed to 257. The Xplained Pro debug tool responds to this with what appears to be the correct data. The stack pointer registers, along with many others, are found at the correct addresses, with the correct values.

I've ran these commands with various memory sizes, and on all occasions, the issue only occurs when requesting more than 256 bytes. But it doesn't *always* occur; I have found that, if I first request <=256 bytes, then issue another request for >256 bytes, both requests return the *correct* data. But if I then issue a third request for >256 bytes, incorrect data is returned. It seems previous requests can affect the validity of subsequent requests. It's very strange, I suspect an issue with the firmware.

Keep in mind that, in both example commands described above, I request no more than the HID report size, so no more than one response packet is expected from the debug tool. I have reviewed the responses from the debug tool, and can confirm it is not attempting to respond with any more than a single packet (the fragment info byte is always `0x11`, one of one).

Do you have any ideas as to what could be happening here? Would appreciate any insight from you.

For now, I have simply implemented a hard limit for the Xplained Pro debug tool. This means that Bloom will never try to request more than 256 bytes of SRAM, from the debug tool, in a single request. It will automatically break down larger requests.

Support for the Xplained Pro debug tool will likely be included in the next release (v0.5.0). This will likely be near the end of December.

mraardvark, 2021-11-22:
Hi @navnavnav 
Interesting observations.  I wonder if this is AVR's repeat-counter being 8-bit?  (Don't have the details off-hand).  (The hard-limit for FW is 900-and something bytes.)
My point regarding the kits: the Xplained kits are considered 'mature' which means that FW bugs will probably not be fixed at this point.  Curiosity Nano is still being developed, and is much easier to debug and fix.  I would consider it wisest to rather start there (too late though) and port/adapt backwards from there.

navnavnav, 2021-11-22:
Hi @mraardvark 

Do you mean the mega4809's repeat-counter, or that of some other AVR on the Xplained Pro board? I ask because I cannot replicate the issue with other debug tools, using the same target; I've tried connecting the Atmel-ICE and the PowerDebugger to the UPDI pins on the Xplained Pro board, and they work fine for reading more than 256 bytes from the target's SRAM.

It's not really much of a problem if the issue isn't fixed. I have a workaround which only slightly impacts performance, so I think we can live with it :)

Ah right, ok, I will purchase a few Curiosity Nano boards over the next few weeks and see if I can get them setup with Bloom. If they're EDBG-based, it shouldn't be too hard. I had the Xplained Pro board working a few hours after creating this issue. It was just that memory reading bug that consumed the weekend. Hopefully the nano board will be more straight-forward. I'll create another issue on GitHub for that 👍🏽 

Thanks for your help, @mraardvark. It is appreciated 👍🏽

mraardvark, 2021-11-22:
@navnavnav - ok, strange - then this sounds like a bug of sorts...
FYI the Atmel-ICE, PowerDebugger and EDBG run our "3G" stack - effectively the same code on all AVR32s, more or less. (which is wierd that EDBG has a bug)
mEDBG - Xplained Mini/Nano is stripped down EDBG to fit on a mega32 (4G)
PICkit4, Snap and nEDBG (Curiosity Nano) all run the "5G" stack - all the same code on ARM debuggers

navnavnav, 2021-11-22:
The implementation for the Xplained Pro debug tool driver is now complete and in the `memory-inspection-insight` branch. I put it in that branch so I can continue testing with the Xplained Pro, as I progress with the [memory inspection ticket](https://github.com/navnavnav/Bloom/issues/7).

As previously stated, it will likely go live with the memory inspection work, in version 0.5.0. I will close this ticket once it's live.

navnavnav, 2021-12-28:
Version 0.5.0 has just been published. See website (https://bloom.oscillate.io) to download latest version and see config docs to select the new debug tool https://bloom.oscillate.io/docs/configuration#debug-tool-target-config.

closed 2021-12-28 by navnavnav