bloombloombloom/Bloom #11: Implement device driver for Curiosity Nano evaluation boards

navnavnav, 2021-11-23:
The Curiosity Nano boards have an on-board debugger which appears to be EDBG-based. I've purchased some of these and hope to begin working on the driver later this week.

Boards I'll be testing with:

- [DM320115](https://uk.farnell.com/microchip/dm320115/curiosity-nano-eval-board-8-bit/dp/2932048?CMP=i-ddd7-00001003)
- [DM080103](https://uk.farnell.com/microchip/dm080103/curiosity-nano-eval-board-8-bit/dp/3219060?CMP=i-ddd7-00001003)
- [DM164151](https://uk.farnell.com/microchip/dm164151/curiosity-nano-dev-board-8-bit/dp/3410690?CMP=i-ddd7-00001003)

navnavnav, 2021-11-27:
Development for this is complete. The EdbgAvr8Interface driver worked without issue, for all three Curiosity Nano evaluation boards.

These changes will be shipped in the next feature release (v0.5.0). Will close this ticket upon release.

ccrause, 2021-12-23:
I am interested in this feature.

navnavnav, 2021-12-28:
Version 0.5.0 has just been published. See website (https://bloom.oscillate.io) to download latest version and see config docs to select the new debug tool https://bloom.oscillate.io/docs/configuration#debug-tool-target-config.

closed 2021-12-28 by navnavnav