bloombloombloom/Bloom #13: AVR8 Activate physical interface command failed - Failure reason: Illegal target ID

ccrause, 2021-12-23:
I'm trying to test Bloom with an MPLAB Snap debugger attached to an attiny24A.  Debugwire fuse is already set on controller, controller powered separately by USB.  Connections:
Snap  - attiny24A
pin 2 - V+
pin 3 - gnd
pin 6 - /reset or dW

Bloom output (bloom.json at end of report):
$ bloom 
2021-12-23 17:10:22 SAST [MT] [1]: [INFO] Selected environment: default
2021-12-23 17:10:22 SAST [TC] [3]: [INFO] Starting TargetController
2021-12-23 17:10:22 SAST [TC] [6]: [INFO] Connecting to debug tool
2021-12-23 17:10:22 SAST [TC] [8]: [INFO] Debug tool connected
2021-12-23 17:10:22 SAST [TC] [9]: [INFO] Debug tool name: MPLAB Snap
2021-12-23 17:10:22 SAST [TC] [10]: [INFO] Debug tool serial: 
2021-12-23 17:10:22 SAST [TC] [13]: [WARNING] AVR8 debugWire interface selected - the DWEN fuse will need to be enabled
2021-12-23 17:10:22 SAST [TC] [14]: [INFO] Activating target
2021-12-23 17:10:23 SAST [TC] [30]: [ERROR] The TargetController encountered a fatal error. See below for errors:
2021-12-23 17:10:23 SAST [TC] [31]: [ERROR] AVR8 Activate physical interface command failed - Failure reason: Illegal target ID
2021-12-23 17:10:23 SAST [TC] [32]: [INFO] Shutting down TargetController
2021-12-23 17:10:23 SAST [TC] [33]: [INFO] Deactivating target
2021-12-23 17:10:23 SAST [TC] [34]: [INFO] Closing debug tool
2021-12-23 17:10:23 SAST [MT] [36]: [ERROR] TargetController failed to startup.
2021-12-23 17:10:23 SAST [MT] [37]: [INFO] Shutting down Bloom
2021-12-23 17:10:23 SAST [SH] [42]: [INFO] Shutting down SignalHandler

I confirmed the device ID read via a different debugwire server (https://github.com/ccrause/debugwire-gdb-bridge):
Device ID: $910B - This ID is the same as the JTAG ID in ATTINY2424A.xml.

Any pointers on how to correctly use Bloom + Snap + debugwire?

bloom.json:
{
  "environments": {
    "default": {
      "debugTool": {
        "name": "snap",
        "releasePostDebugSession": true
      },

      "target": {
        "name": "attiny24A",
        "physicalInterface": "debug-wire"
      },

      "debugServer": {
        "name": "avr-gdb-rsp",
        "ipAddress": "127.0.0.1",
        "port": "1234"
      }
    }
  },

  "insight": {
    "enabled": true
  }
}

navnavnav, 2021-12-23:
HI @ccrause, thanks for reporting this.

Hmm, your bloom config seems fine. And the error you're seeing is from the SNAP itself. Do you have anything (besides the SNAP) connected to your dW/RESET pin? Any pull-ups or capacitors? If so, try removing those if you can.

What about lock bits - have you programmed any of those? They should be unprogrammed before you can use the dW module on the device.

See the "Physical Interface" subsection in section 17, in the datasheet, for more: https://ww1.microchip.com/downloads/en/DeviceDoc/ATtiny24A-44A-84A-DataSheet-DS40002269A.pdf

Failing the above, try using an MPLAB tool like the MPLAB IPE to connect to the device. Can you read any target memories via that? Can you send over the value for the three fuse bytes (high, low and extended).

Also, please try enabling Bloom's debug logging by setting `debugLoggingEnabled` to true, in your bloom.json. The `debugLoggingEnabled` should be specified at the project level:

```
{
  "debugLoggingEnabled": true,

  "environments": {...},

  "debugServer": {... },

  "insight": {... }
}
```

Then run Bloom again and send over the output.

ccrause, 2021-12-23:
I have a 10k pullup on reset, but can read SRAM in MPLAB X IDE using Snap, so the Snap <-> Attiny24A communication appears OK.  I cannot pull the fuses over debugwire, but know that the debugwire and internal 8MHz oscillator options are set.  No lock bits are set, device can be programmed over debugwire.

Debug log below suggest there is a missing SIGNATURE BASE ADDRESS.

$ bloom 
2021-12-23 20:51:30 SAST [MT] [1]: [DEBUG] Debug log printing has been enabled.
2021-12-23 20:51:30 SAST [MT] [2]: [INFO] Selected environment: default
2021-12-23 20:51:30 SAST [MT] [3]: [DEBUG] Number of environments extracted from config: 1
2021-12-23 20:51:30 SAST [SH] [4]: [DEBUG] Starting SignalHandler
2021-12-23 20:51:30 SAST [SH] [5]: [DEBUG] SignalHandler ready
2021-12-23 20:51:30 SAST [TC] [6]: [INFO] Starting TargetController
2021-12-23 20:51:30 SAST [TC] [7]: [INFO] Connecting to debug tool
2021-12-23 20:51:30 SAST [TC] [8]: [DEBUG] HID device path: 0001:0022:00
2021-12-23 20:51:30 SAST [TC] [9]: [INFO] Debug tool connected
2021-12-23 20:51:30 SAST [TC] [10]: [INFO] Debug tool name: MPLAB Snap
2021-12-23 20:51:30 SAST [TC] [11]: [INFO] Debug tool serial: 
2021-12-23 20:51:30 SAST [TC] [12]: [DEBUG] Loading AVR8 target description file: /opt/bloom/bin/../resources/TargetDescriptionFiles/AVR/AVR8/TINY/ATTINY24A.xml
2021-12-23 20:51:30 SAST [TC] [13]: [DEBUG] Failed to extract pinout from target description element - Failed to convert position attribute value to integer on pin element 0
2021-12-23 20:51:30 SAST [TC] [14]: [INFO] Activating target
2021-12-23 20:51:30 SAST [TC] [15]: [ERROR] The TargetController encountered a fatal error. See below for errors:
2021-12-23 20:51:30 SAST [TC] [16]: [ERROR] Missing required parameter: SIGNATURE BASE ADDRESS
2021-12-23 20:51:30 SAST [TC] [17]: [INFO] Shutting down TargetController
2021-12-23 20:51:30 SAST [TC] [18]: [INFO] Deactivating target
2021-12-23 20:51:30 SAST [TC] [19]: [INFO] Closing debug tool
2021-12-23 20:51:30 SAST [TC] [20]: [DEBUG] Event "TargetControllerThreadStateChanged" (1) registered for listener ApplicationEventListener
2021-12-23 20:51:30 SAST [MT] [21]: [ERROR] TargetController failed to startup.
2021-12-23 20:51:30 SAST [MT] [22]: [INFO] Shutting down Bloom
2021-12-23 20:51:30 SAST [MT] [23]: [DEBUG] Joining TargetController thread
2021-12-23 20:51:30 SAST [MT] [24]: [DEBUG] TargetController thread joined
2021-12-23 20:51:30 SAST [MT] [25]: [DEBUG] Joining SignalHandler thread
2021-12-23 20:51:30 SAST [SH] [26]: [DEBUG] SIGNAL 10 received
2021-12-23 20:51:30 SAST [SH] [27]: [INFO] Shutting down SignalHandler
2021-12-23 20:51:30 SAST [MT] [28]: [DEBUG] SignalHandler thread joined

navnavnav, 2021-12-23:
@ccrause Hmm, that's a different error now. It looks to be a result of missing information in the target description file (`/opt/bloom/resources/TargetDescriptionFiles/AVR/AVR8/TINY/ATTINY24A.xml`). Have you made any changes to that file? If so, please revert and send over the log with the previous error. If you've not made any changes, something really weird is happening there and I'll need to investigate further. I have scripts in place to ensure that all TDFs are valid before they're shipped, so not sure what's going on there.

If you didn't make any changes to the TDF, please send over the contents of that file for examining.

Is there any chance you can remove that pullup on the reset line? Do you have another target you can try with? I don't have a tiny24a, so I can't test on my end. But I have a mega328p and that works fine with dW and the Snap. 

Also, one other thing to try is to set your target config to `avr8`. This should result in an "ambiguous signature error", but for that to happen, Bloom must be able to extract the signature from the target. So this will give me an idea to whether Bloom is able to talk to the target at all.

Please send over the output.

ccrause, 2021-12-23:
Doh! In all my testing I changed physicalInterface to updi, hence the error.  Changed it back to debug-wire and now Bloom can start up the debug server.

Not sure what caused the first error though, possibly some issue with the config of Snap, since the only change I made since my first test was to read the attiny24A in MPLAB X over Snap.

Apologies for the noise, and thank you for the help!

closed 2021-12-23 by ccrause

navnavnav, 2021-12-23:
Ah yes, that would explain the second error.

Glad you got it working 👍🏽 

Let me know if you have any other issues.