bloombloombloom/Bloom #14: Gdbserver registers response shorter than gdb expects

ccrause, 2021-12-23:
Testing Bloom with gdb 10.2 resulted in gdb complaining the response to the 'g' request is truncated.  Inspection shows that the response is one byte shorter than the expected length of 32+1+2+4 = 39 bytes (or 78 hex digits).  Please find the gdb session below.

$ ~/gdb/avr-gcc-11.1.0-x64-linux/bin/avr-gdb
GNU gdb (GDB) 10.2
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=avr".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) set debug remote 1
(gdb) fil testbreak.X.debug.elf 
Reading symbols from testbreak.X.debug.elf...
(gdb) tar rem :1234
Remote debugging using :1234
Sending packet: $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df...Ack
Packet received: qSupported:swbreak+;PacketSize=1024;
Packet qSupported (supported-packets) is supported
Sending packet: $vMustReplyEmpty#3a...Ack
Packet received: \000
Sending packet: $Hg0#df...Ack
Packet received: \000
Sending packet: $qTStatus#49...Ack
Packet received: \000
Packet qTStatus (trace-status) is NOT supported
Sending packet: $?#3f...Ack
Packet received: T05
Sending packet: $qfThreadInfo#bb...Ack
Packet received: \000
Sending packet: $qL1200000000000000000#50...Ack
Packet received: \000
Sending packet: $Hc-1#09...Ack
Packet received: \000
Sending packet: $qC#b4...Ack
Packet received: \000
Sending packet: $qAttached#8f...Ack
Packet received: \001
Packet qAttached (query-attached) is supported
Sending packet: $qOffsets#4b...Ack
Packet received: \000
Sending packet: $g#67...Ack
Packet received: ff00ffffbffffffd7fdffffdbeffbfffffffffffffbfefff20000301df08370000df00000000
Truncated register 34 in remote 'g' packet
(gdb) info registers
The program has no registers now.

navnavnav, 2021-12-23:
Can you attach the debug log please?

ccrause, 2021-12-23:
Certainly, Bloom log:

$ bloom 
2021-12-23 22:50:23 SAST [MT] [1]: [DEBUG] Debug log printing has been enabled.
2021-12-23 22:50:23 SAST [MT] [2]: [INFO] Selected environment: default
2021-12-23 22:50:23 SAST [MT] [3]: [DEBUG] Number of environments extracted from config: 1
2021-12-23 22:50:23 SAST [SH] [4]: [DEBUG] Starting SignalHandler
2021-12-23 22:50:23 SAST [SH] [5]: [DEBUG] SignalHandler ready
2021-12-23 22:50:23 SAST [TC] [6]: [INFO] Starting TargetController
2021-12-23 22:50:23 SAST [TC] [7]: [INFO] Connecting to debug tool
2021-12-23 22:50:23 SAST [TC] [8]: [DEBUG] HID device path: 0001:0022:00
2021-12-23 22:50:23 SAST [TC] [9]: [INFO] Debug tool connected
2021-12-23 22:50:23 SAST [TC] [10]: [INFO] Debug tool name: MPLAB Snap
2021-12-23 22:50:23 SAST [TC] [11]: [INFO] Debug tool serial: 
2021-12-23 22:50:23 SAST [TC] [12]: [DEBUG] Loading AVR8 target description file: /opt/bloom/bin/../resources/TargetDescriptionFiles/AVR/AVR8/TINY/ATTINY24A.xml
2021-12-23 22:50:23 SAST [TC] [13]: [DEBUG] Failed to extract pinout from target description element - Failed to convert position attribute value to integer on pin element 0
2021-12-23 22:50:23 SAST [TC] [14]: [WARNING] AVR8 debugWire interface selected - the DWEN fuse will need to be enabled
2021-12-23 22:50:23 SAST [TC] [15]: [INFO] Activating target
2021-12-23 22:50:23 SAST [TC] [16]: [DEBUG] Setting DEVICE_FLASH_PAGE_SIZE AVR8 parameter
2021-12-23 22:50:23 SAST [TC] [17]: [DEBUG] Setting DEVICE_FLASH_SIZE AVR8 parameter
2021-12-23 22:50:23 SAST [TC] [18]: [DEBUG] Setting DEVICE_FLASH_BASE AVR8 parameter
2021-12-23 22:50:23 SAST [TC] [19]: [DEBUG] Setting DEVICE_SRAM_START AVR8 parameter
2021-12-23 22:50:23 SAST [TC] [20]: [DEBUG] Setting DEVICE_EEPROM_SIZE AVR8 parameter
2021-12-23 22:50:23 SAST [TC] [21]: [DEBUG] Setting DEVICE_EEPROM_PAGE_SIZE AVR8 parameter
2021-12-23 22:50:23 SAST [TC] [22]: [DEBUG] Setting DEVICE_OCD_REVISION AVR8 parameter
2021-12-23 22:50:23 SAST [TC] [23]: [DEBUG] Setting DEVICE_OCD_DATA_REGISTER AVR8 parameter
2021-12-23 22:50:23 SAST [TC] [24]: [DEBUG] Setting DEVICE_SPMCR_REGISTER AVR8 parameter
2021-12-23 22:50:23 SAST [TC] [25]: [DEBUG] Setting DEVICE_OSCCAL_ADDR AVR8 parameter
2021-12-23 22:50:23 SAST [TC] [26]: [DEBUG] Setting DEVICE_EEARL_ADDR AVR8 parameter
2021-12-23 22:50:23 SAST [TC] [27]: [DEBUG] Setting DEVICE_EEARH_ADDR AVR8 parameter
2021-12-23 22:50:23 SAST [TC] [28]: [DEBUG] Setting DEVICE_EECR_ADDR AVR8 parameter
2021-12-23 22:50:23 SAST [TC] [29]: [DEBUG] Setting DEVICE_EEDR_ADDR AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [30]: [INFO] Target activated
2021-12-23 22:50:24 SAST [TC] [31]: [INFO] AVR8 target promoted to tinyAVR target
2021-12-23 22:50:24 SAST [TC] [32]: [DEBUG] Setting DEVICE_FLASH_PAGE_SIZE AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [33]: [DEBUG] Setting DEVICE_FLASH_SIZE AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [34]: [DEBUG] Setting DEVICE_FLASH_BASE AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [35]: [DEBUG] Setting DEVICE_SRAM_START AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [36]: [DEBUG] Setting DEVICE_EEPROM_SIZE AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [37]: [DEBUG] Setting DEVICE_EEPROM_PAGE_SIZE AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [38]: [DEBUG] Setting DEVICE_OCD_REVISION AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [39]: [DEBUG] Setting DEVICE_OCD_DATA_REGISTER AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [40]: [DEBUG] Setting DEVICE_SPMCR_REGISTER AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [41]: [DEBUG] Setting DEVICE_OSCCAL_ADDR AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [42]: [DEBUG] Setting DEVICE_EEARL_ADDR AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [43]: [DEBUG] Setting DEVICE_EEARH_ADDR AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [44]: [DEBUG] Setting DEVICE_EECR_ADDR AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [45]: [DEBUG] Setting DEVICE_EEDR_ADDR AVR8 parameter
2021-12-23 22:50:24 SAST [TC] [46]: [INFO] Target ID: 0x1e910b
2021-12-23 22:50:24 SAST [TC] [47]: [INFO] Target name: ATtiny24A
2021-12-23 22:50:24 SAST [TC] [48]: [DEBUG] Event "TargetControllerThreadStateChanged" (2) registered for listener ApplicationEventListener
2021-12-23 22:50:24 SAST [TC] [49]: [DEBUG] TargetController ready and waiting for events.
2021-12-23 22:50:24 SAST [MT] [50]: [INFO] Selected DebugServer: AVR GDB Remote Serial Protocol Debug Server
2021-12-23 22:50:24 SAST [DS] [51]: [INFO] Starting DebugServer
2021-12-23 22:50:24 SAST [TC] [52]: [DEBUG] Target state changed - STOPPED
2021-12-23 22:50:24 SAST [DS] [53]: [DEBUG] Event "ExtractTargetDescriptor" (3) registered for listener TargetControllerEventListener
2021-12-23 22:50:24 SAST [TC] [54]: [DEBUG] Dispatching event ExtractTargetDescriptor (3).
2021-12-23 22:50:24 SAST [TC] [55]: [DEBUG] Event "TargetDescriptorExtracted" (5) registered for listener DebugServerEventListener
2021-12-23 22:50:24 SAST [DS] [56]: [INFO] GDB RSP address: 127.0.0.1
2021-12-23 22:50:24 SAST [DS] [57]: [INFO] GDB RSP port: 1234
2021-12-23 22:50:24 SAST [DS] [58]: [DEBUG] Event "DebugServerThreadStateChanged" (6) registered for listener ApplicationEventListener
2021-12-23 22:50:24 SAST [DS] [59]: [INFO] DebugServer ready
2021-12-23 22:50:24 SAST [DS] [60]: [INFO] Waiting for GDB RSP connection
2021-12-23 22:50:24 SAST [DS] [61]: [DEBUG] GDB RSP interrupted
2021-12-23 22:50:24 SAST [DS] [62]: [INFO] Waiting for GDB RSP connection
2021-12-23 22:50:25 SAST [MT] [63]: [INFO] Starting Insight
[... cut ...]
2021-12-23 22:50:45 SAST [TC] [146]: [DEBUG] Event "TargetControllerStateReported" (19) registered for listener InsightWorkerEventListener
2021-12-23 22:50:45 SAST [TC] [147]: [DEBUG] TargetController suspended
2021-12-23 22:50:45 SAST [DS] [148]: [DEBUG] GDB RSP interrupted
2021-12-23 22:50:45 SAST [DS] [149]: [DEBUG] Dispatching event TargetControllerStateReported (19).
2021-12-23 22:50:45 SAST [DS] [150]: [INFO] Waiting for GDB RSP connection
2021-12-23 22:50:45 SAST [IW] [151]: [DEBUG] Dispatching event TargetControllerStateReported (19).

navnavnav, 2021-12-23:
OK, I think I see the problem here. Your target only has a single byte SP register, and Bloom is sending it as is.

I can sort a fix for this, but it will be shipped in the next release, which is due late next week. Will keep you updated via this ticket.

ccrause, 2021-12-23:
Thank you for investigating. Will wait for the next release.

navnavnav, 2021-12-28:
Version 0.5.0 has just been published. It contains a fix for this issue. See website (https://bloom.oscillate.io) to download the latest version. As I don't have an any targets with a single byte SP (there are only 9 of the 249 supported targets) to test with, it would be good if you could test and update this ticket after.

I will close this ticket once I've received confirmation from you.

ccrause, 2021-12-28:
Thanks Nav.  I confirm that this issue with gdb is now fixed as tested with an attiny24A over Snap.

closed 2021-12-28 by ccrause