bloombloombloom/Bloom #15: Missing support for attiny45

ccrause, 2021-12-24:
The next controller I had ready for testing was an attiny45, which is not supported.  I though I could just copy the attiny45.atdf file, change the extensions and the name of the top node, but that didn't work (target name not recognised).

navnavnav, 2021-12-27:
Thanks for reporting this, @ccrause.

I've added a TDF for the ATtiny45. This will be shipped with v0.5.0 later this week. Will close this ticket upon release.

navnavnav, 2021-12-28:
Version 0.5.0 has just been published. See website (https://bloom.oscillate.io) to download the latest version. As I don't have an Attiny45 to test with, it would be good if you could test and update this ticket after.

I will close this ticket once I've received confirmation from you.

ccrause, 2021-12-28:
Thank you Nav.  I can confirm that Bloom can now connect successfully to an attiny45 via Snap.

closed 2021-12-28 by ccrause