bloombloombloom/Bloom #16: Uploading code when debugWIRE is enabled

lkoepsel, 2021-12-26:
I've just started working with Bloom and I am very impressed! The insight dashboard is lovely and it is very helpful to set the registers so easily! 

I'm using Bloom and a Snap to replace avarice and an Atmel Dragon. I write curriculum for college students and I'm using the ATmega328p as the standard processor. Having a more modern tool like Bloom is great! I don't use CLion, I use avr-gdb --tui to keep things simple. 

That said, the Dragon is able to program the 328 via avrdude when debugWIRE is enabled. I haven't been able to perform the same task using a Bloom/Snap/avrdude combination. 

I can upload code via MPLAB IPE as it will turn off debugWIRE, however, then I need to use avrdude and the Dragon to turn debugWIRE back on.

Any thoughts?

Thanks,
Lief

lkoepsel, 2021-12-26:
As an update, I also implemented the UPDI fix advised by Microchip. I removed the resistor R48 and added a 5k resistor in between VTG and MOSI (SIL pin 2-4). No change to outcomes.

navnavnav, 2021-12-26:
Hi Leif,

Thanks for raising this with me.

What version of avrdude are you using? The one I'm using (v6.2) temporarily disables the debugWire module, then proceeds with programming. Bloom will continue to work, but only after power to the target has been cycled. This is because the debugWire module will remain temporarily disabled until the power is cycled.

If I remember correctly, I had installed a newer version of avrdude, but I downgraded shortly after. There was an issue with the more recent version, where it would fail to disable the debugWire module and so it would fail to program the target. In some cases, it even left the debug tool in a bad state, where a hard-reset was required for continued use. This, if I remember correctly, was the reason for downgrading to v6.2. But this was some time ago, so it's possible a newer version would have addressed this issue. And I'm afraid downgrading won't be an option for you, as v6.2 doesn't appear to support the SNAP (I was using the Atmel-ICE at the time).

There may be something I can do to help with this;

Bloom possesses the ability to temporarily disable the debugWire module. This is something I had implemented before the initial release, but I left it disabled as it seemed avrdude (and I assumed MPLAB IPE) would take care of it.
I've just tried to enable this on my local build, and I can confirm that it still works, and that MPLAB IPE can program the target, whilst debugWire is temporarily disabled. And the programming does not result in the DWEN fuse being reset. I can also confirm that Bloom still works fine after programming, but a power cycle is still required after each instance of programming.

So, I propose the following:

1. I will enable the function to temporarily disable the debugWire module, in Bloom. I will make this optional via a target configuration parameter that will be set in your project configuration file (bloom.json). When enabled, Bloom will automatically disable the debugWire module just before disconnecting from the target.
2. With this function enabled, you will be able to program the target via ISP and MPLAB IPE. This should **not** result in the debugWire module being completely disabled (via the resetting of the DWEN fuse).
3. Once programming has finished, you will be required to cycle the power to the target, in order to enable the debugWire module. From this point on wards, you should be able to continue using Bloom.

Drawbacks to this approach:
1. You'll have to cycle the power to the target after each instance of programming. There's no avoiding this.
2. You'll likely have/want to disable the `releasePostDebugSession` option in Bloom. With this option enabled, Bloom will disable the debugWire module every time a debug session comes to an end. This will likely result in frustration when the user finishes a debug session with no intention of programming the target, then attempts to start a new debug session, only for Bloom to fail to connect to the target (due to the debugWire module being temporarily disabled). Disabling the `releasePostDebugSession` function will mean Bloom will only disable the debugWire module when the user quits the application (giving the user more control of when the debugWire module is temporarily disabled).

Let me know if you wish to proceed with this approach. I may be able to get the changes into the next release, which is due by next Sunday. 

Another option would be to consider using a different, non-debugWire target. Preferably one with UPDI support. But keep in mind that there is a known interfacing issue with the SNAP and UPDI targets. The solution is simple - all it requires is a single pull-up resistor on the UPDI line. But still worth keeping in mind.

navnavnav, 2021-12-26:
Thinking about this some more, another approach would be to implement the ISP interface in Bloom. This would allow Bloom to force the enabling of the debugWire module, via ISP. Then it could switch back to the AVR8 generic protocol and proceed with the debug session. I suspect this is what the MPLAB IDE does.

I'll create a new ticket for this, but it's a lot of work and it won't be ready until the next feature release (a number of months from now).

lkoepsel, 2021-12-26:
Thanks for the response so quickly! 

Again, I can't tell you enough, how impressed I am with this software. It is exactly what I was looking for. BTW - it was your instructions on how to set AVR mode on the Snap which got me going back in this direction! Thanks!

More on my work with the Snap...
1) Yes, I am using 6.4 as it is the version which enables the Snap to work. And you are correct, downgrading would remove that capability. I have ordered an Atmel ICE, however my goal is to help students understand the least expensive way to do this and the Snap is clearly the best low cost solution (for both debugging and programming the 328P).
2. I was able to use MPLAB IDE to program the board without disabling debugWIRE. Here is my process:
   a) Start Bloom and keep it running
   b) Start MPLAB IDE and keep it running
   c) In another terminal window, run make to compile and create a .hex file
   d) Go into IDE and refresh/reload the .hex file in the project 
   e) In IDE, Production -> Program Device for Debugging Main Project
   f) Confirm Bloom is still running (it is!!)
   g) Go back to debugging using avr-gdb

I think everything is working "OK" for now.  When you have the chance to "implement ISP"...that would be fantastic! I can explain this method easy enough. I've repeated these steps multiple times and they continue to work.

Does this make sense?

Thanks!
Lief

lkoepsel, 2021-12-27:
I continued to work with the process above. I have some wonky errors, where avr-gdb crashes when I load a new file. I'm not sure if that is user error or software.

It's the end of my day...I'll pick this back up in the morning, Bloom has been a great find!!

navnavnav, 2021-12-27:
Hi Leif,

It's good to hear that the documentation helped. And thank you for your kinds words RE Bloom.

I completely understand RE the Atmel-ICE. It's an excellent debug tool, and you won't have this issue with it as avrdude supports it. But may not be the best option for students, given the price tag.

The steps you've detailed make sense. I suspect the MPLABX IDE is doing what I suggested Bloom do with the ISP interface; It first attempts to connect via debugWire, then fails due to debugWire being disabled, so fallsback to the ISP interface. With ISP, the IDE then enables debugWire and makes another connection attempt.

So it's a good workaround, but not ideal TBH. The whole point of Bloom is to allow users to move away from relying on vendor supplied IDEs.

Fortunately, I've managed to make those changes I suggested in my previous comment, RE Bloom disabling the debugWire module pre-disconnect. They'll be shipped in the next release some time later this week. With those changes, you can simply set the new `disableDebugWirePreDisconnect` option in your Bloom config:
```
"target": {
    "name": "atmega328p",
    "physicalInterface": "debug-wire",
    "disableDebugWirePreDisconnect": true
}
```
And disable `releasePostDebugSession`:
```
"debugTool": {
    "name": "snap",
    "releasePostDebugSession": false
}
```
Then, you can leave Bloom running up until the point when you wish to program the device. When you're ready to program, simply close Bloom to have it temporarily disable the debugWire module. Then MPLAB IPE will be able to program it without that warning RE disabling the debugWire module. Once you're done programming, simply cycle the power to the target and run Bloom again.

It's still a bit tedious but at least you won't have to use the MPLAB IDE or avrdude to enable debugWire, each time you program the device.

What would be really nice is if the IPE could put the device back into debugWire mode, upon the completion of programming. Might be worth raising that with Microchip support.

lkoepsel, 2021-12-27:
Two things:
1) Thank you! The fix makes sense and I look forward to testing it!
2) Yes, regarding the Microchip issue, I was surprised that the IPE would be so quick to set the config bit from debugWIRE to ISP, however, not set it back. Its an extremely common software practice. As my father used to say...(he was a former PhD in EE)...put things back where you found them. :)

Lief

mraardvark, 2021-12-27:
Stupid question perhaps (and I haven't really been following this discussion properly): if you don't need to adjust fuses, why not just use the AVR8 protocol to program flash when starting a debugWIRE session?

ccrause, 2021-12-27:
> Thinking about this some more, another approach would be to implement the ISP interface in Bloom. This would allow Bloom to force the enabling of the debugWire module, via ISP. Then it could switch [...]

I would also like this feature!  It is really unfortunate that flash writes are not allowed by EDBG/AVR8 protocol while debugging, since it is possible over the plain debugwire protocol.

mraardvark, 2021-12-27:
>  It is really unfortunate that flash writes are not allowed by EDBG/AVR8 protocol while debugging
Of course writing flash is allowed, just not actually _during_ the session, since this would wreak havoc with the SWBPs.

navnavnav, 2021-12-27:
@mraardvark Bloom doesn't currently support programming the target. This is something that I've considered in the past, but have decided against it because A. I want to avoid Bloom becoming an "all in one" solution for programming and debugging - it would be too much of a strain on me to support. B. It would be reinventing the wheel, given that there are already two excellent options (IPE and avrdude) for programming targets, already available to Linux users.

The problem here is that the user is attempting to program the target via avrdude/IPE, but both of these appear to only be able to program the device via ISP. I can't see a way to configure IPE to use debugWire for programming:

[image]

The only option there is ISP. If IPE could program via debugWire, that would solve this problem and would be excellent.

mraardvark, 2021-12-27:
OK, I appreciate that view... (but :)
debugWIRE has always been a bit of an odd interface because its not conducive to separation of programming and debugging in that 'traditional' way given the DWEN fuse issue.
Microchip will never add debugWIRE to the IPE because debugWIRE is "not a programming interface".  Arguably programming flash by debugWIRE is considered part of a debugging operation.  In fact you have already configured all you need in the AVR8 protocol at this point...

navnavnav, 2021-12-27:
Thanks for clarifying, @mraardvark 

I have created a ticket for implementing support for the ISP protocol: https://github.com/navnavnav/Bloom/issues/17
Will post any updates on there, as and when necessary.

I will close this ticket once v0.5.0, containing the partial solution, has been published.

navnavnav, 2021-12-28:
Version 0.5.0 has just been published. See website (https://bloom.oscillate.io) to download latest version and see config docs for info on the new target parameter.

closed 2021-12-28 by navnavnav