bloombloombloom/Bloom #17: Implement the ISP interface for a better experience with debugWire

navnavnav, 2021-12-27:
Currently, Bloom can only interface with AVR8 targets using the AVR8 Generic EDBG protocol.

For targets with a debugWire module, a fuse known as "DWEN" (debugWire enable) must be set in order to activate the module and perform debugging operations on the target. This fuse cannot be set via the AVR8 Generic protocol.

Because Bloom is unable to set this fuse, responsibility to set the fuse and (ensure that it remains set) lies with the user. This means the user has to set the fuse each time after programming the target via ISP (if they wish to perform debugging operations).

AVRDUDE used to help with this by temporarily disabling the debugWire module, before employing the ISP protocol to interface with the target. This meant that the user could simply reactivate the debugWire module by cycling the power to the target. But this function in AVRDUDE appears to no longer work (see https://github.com/navnavnav/Bloom/issues/16 for more on that).

In an ideal world, Bloom should be able to control the state of the debugWire module without relying on the use of other software. This would result in a seamless user experience with debugWire targets. It can be done via the implementation of the ISP protocol. With this implementation, Bloom will be able to connect to the target via SPI and set/clear the DWEN fuse automatically.

Not sure when I'll be able to start work on this. Will update the ticket as and when necessary.

lkoepsel, 2021-12-29:
Version 0.5 is looking very good! I haven't had a chance to review the memory feature, as I wanted to lock down the Snap/debugWIRE programming process based on this new release:

First I test the original premise from issue #16 which is:
1. Add  "disableDebugWirePreDisconnect": true to the bloom.json file
2. Close bloom and avr-gdb when you wish to program the chip
3. Use MPE IPE to program
4. Cycle power
5. Restart Bloom then avr-gdb
Not ideal, however, it works and that is what is important.

My first attempt to improve was to replace IPE with avrdude, as avrdude can be added to a makefile script.
By following this link: 
[AVRFreaks: MPLAB SNAP #120](https://www.avrfreaks.net/comment/3070341#comment-3070341)
I added a snap_isp configuration to avrdude.conf
This worked! And is a much better solution than IPE.

This means my new process is:
1. Close bloom and avr-gdb
2. make
3. Cycle power
4. Open bloom and avr-gdb

I've also eased the power cycling by adding a simple USB hub with power switch like this one: [USB Hub with Switch](https://www.adafruit.com/product/2991)

Thank you for making this successful! 

Now let's see how to short-term improve:
1) Is there a way to programmatically close bloom and avr-gdb (or have bloom close when avr-gdb closes?) I could add this to my makefile to shorten steps 1 & 2.

I am also going to explore using a ATmega328PB XPLAINED as a student's test board, I believe you just added it to the list of interfaces.

navnavnav, 2021-12-29:
Hi Lief,

That's great news RE the new target param working as expected.

To kill Bloom problematically, you can just use `pkill`
```
pkill -2 Bloom
```
The `-2` sends the SIGINT signal to Bloom, allowing Bloom a chance to shutdown gracefully (which is where it attempts to disable the debugWire module).

And you can kill avrdude with `pkill` also:
```
pkill -2 avrdude
```

navnavnav, 2021-12-29:
Sorry, I misread your previous comment. Thought you wanted to kill avrdude with Bloom. See edited version for killing avr-gdb. 👍🏽

lkoepsel, 2021-12-30:
I worked a bit on attempting to automate quitting bloom and gdb, the former does quite nicely and the latter is a bit of an issue. I'm going to leave that aspect alone for now.

I wanted to mention a situation which happened this morning for others to understand how to fix. 

I believed I had left my Uno/Snap/System in a good state last night, however, when I began to work on it this morning I had no luck getting avrdude to work with the Uno. I attempted to start/stop Bloom, restart my system, however each time, avrdude responded with various "unexpected response", "content mismatch" etc type errors.

After trying multiple methods of trying to get the Uno back to a known state, I went back to MPLAB IPE. It too had issues, however, it responded with a *"A communication error with the debug tool has occurred. The tool will be reset and should re-enumerate shortly. Connection Failed."*
It then attempted to reconnect, it did so successfully and programmed the Uno. I did a little bit of testing the connection with Bloom and avr-gdb and everything was working as expected.

I then edited my file, reattempted to "quit avr-gdb/quit bloom/make etc/power cycle" and everything worked! avrdude had no issues and ran as expected.

My lesson is *"don't delete MPLAB IPE, as you might need it to get everything back into a known state"*

navnavnav, 2021-12-30:
Hi Lief,

I've had that very same issue. Are you still using the newer version of avrdude? For me, it was that version of avrdude that was leaving the debug tool in a bad state. Did you try disconnecting the debug tool and reconnecting? That's effectively what IPE would have done judging by that error message.

If it happens again, try disconnecting power to the snap. Be sure to disconnect both from USB and from your external power supply. Then reconnect and try again.

lkoepsel, 2021-12-30:
I am using averdude 6.4, I was under the belief that averdude 6.2 wouldn't work with Snap. I thought I had disconnected the Snap, however, I might not have. Makes sense as to why IPE "fixed" it.

One thing I did change was only dropping power to the target instead of both the target and the Snap. That might have been an issue as well.

Quitting avr-gdb wasn't as much of an issue as adding it to the automation. It made more sense for me to be in avr-gdb, quit it manually, then stay in the same window running a string of commands to kill bloom/make etc.

When it happens again, I'll spend a bit more time debugging knowing I have some things to try. :)

BTW - I'm setting up three pages on my site for using Bloom with avr-gdb, Setup, Examples and Hints. This issue would fall into the Hints (meaning when all else fails, try one of these Hints.)  I'll have an early version of them by end of day today (Dec 30).

navnavnav, 2022-03-02:
Quick update on this:

I've made good progress. The implementation for the AVRISP EDBG protocol is almost complete. I have successfully established a connection with an AVR target and extracted the target's signature and fuse bytes, via the SPI interface.

I'm hoping to have this ready for testing in the next week or two. I've only been testing with the Atmel-ICE, so I will likely find some issues when I begin testing with other debug tools.

Still looking to have this in the next release (0.7.0). Not sure when that will be, but likely this month (March).

lkoepsel, 2022-03-03:
I look forward to testing! 
Thanks for the update!
Lief

lkoepsel, 2022-03-09:
The situation certainly sounds familiar! 

I have "left it working" then come back to find out it doesn't along with the errors you describe. I believe my "fix" has been to fall back to Microchips IPE to reset the board. 

Which ironically led me to Bloom...(on one of my searches to fix the problem).

Nice work on solving the issue!

navnavnav, 2022-03-15:
Oh and BTW @lkoepsel, v0.7.0 also introduces the new `shutdownPostDebugSession` option. Bloom can now be configured to automatically shutdown at the end of a debug session. See https://bloom.oscillate.io/docs/configuration#debug-session-config for more.

lkoepsel, 2022-03-15:
I'm having issues, however, I'm not sure if it is user or a bug...here is the situation:
## hardware
* Microchip Snap
* Arduino Uno
## bloom.json
```json
{
  "environments": {
    "default": {
      "shutdownPostDebugSession": true,
      "debugTool": {
        "name": "snap",
        "releasePostDebugSession": false
      },

      "target": {
        "name": "atmega328p",
        "physicalInterface": "debug-wire",
        "disableDebugWirePreDisconnect": false,
         "updateDwenFuseBit": true
      },

      "debugServer": {
        "name": "avr-gdb-rsp",
        "ipAddress": "127.0.0.1",
        "port": "1442"
      }
    }
  },

  "insight": {
    "enabled": true
  }
}
```
## Scenario
1. I start with a working ISP link where I can execute the commands *make clean && make flash_snap*, where *flash_snap* is using avrdude 6.4 and snap_isp mode.
2. I startup bloom:
```bash
(.pyutil) lkoepsel@manjaroMBP Documents/AVR_C (main *%) » bloom
2022-03-15 10:29:18 PDT [MT] [2]: [INFO] Selected environment: "default"
2022-03-15 10:29:18 PDT [TC] [6]: [INFO] Starting TargetController
2022-03-15 10:29:18 PDT [TC] [7]: [INFO] Connecting to debug tool
2022-03-15 10:29:18 PDT [TC] [9]: [INFO] Debug tool connected
2022-03-15 10:29:18 PDT [TC] [10]: [INFO] Debug tool name: MPLAB Snap
2022-03-15 10:29:18 PDT [TC] [11]: [INFO] Debug tool serial: 
2022-03-15 10:29:19 PDT [TC] [13]: [WARNING] AVR8 debugWire interface selected - the DWEN fuse will need to be enabled
2022-03-15 10:29:19 PDT [TC] [14]: [INFO] Activating target
2022-03-15 10:29:22 PDT [TC] [31]: [WARNING] Failed to activate the debugWire physical interface - attempting to access target via the ISP interface, for DWEN fuse bit inspection.
2022-03-15 10:29:22 PDT [TC] [33]: [INFO] Initiating ISP interface
2022-03-15 10:29:23 PDT [TC] [34]: [WARNING] Updating the DWEN fuse bit is a potentially dangerous operation. Bloom is provided "AS IS", without warranty of any kind. You are using Bloom at your own risk. In no event shall the copyright owner or contributors be liable for any damage caused as a result of using Bloom. For more details, see the Bloom license at https://bloom.oscillate.io/license
2022-03-15 10:29:23 PDT [TC] [35]: [INFO] Reading target signature via ISP
2022-03-15 10:29:23 PDT [TC] [36]: [INFO] Target signature confirmed: 0x1e950f
2022-03-15 10:29:23 PDT [TC] [37]: [INFO] Current SPIEN fuse bit value confirmed
2022-03-15 10:29:23 PDT [TC] [38]: [INFO] Cleared lock bits confirmed
2022-03-15 10:29:23 PDT [TC] [39]: [WARNING] Programming DWEN fuse bit
2022-03-15 10:29:23 PDT [TC] [40]: [INFO] DWEN fuse bit successfully updated
2022-03-15 10:29:23 PDT [TC] [41]: [INFO] Retrying debugWire physical interface activation
2022-03-15 10:29:26 PDT [TC] [43]: [ERROR] The TargetController encountered a fatal error. See below for errors:
2022-03-15 10:29:26 PDT [TC] [44]: [ERROR] Failed to activate the debugWire physical interface - check target connection. If the target was recently programmed via ISP, try cycling the target power. See https://bloom.oscillate.io/docs/debugging-avr-debugwire for more information.
2022-03-15 10:29:26 PDT [TC] [45]: [INFO] Shutting down TargetController
2022-03-15 10:29:26 PDT [TC] [46]: [INFO] Deactivating target
2022-03-15 10:29:26 PDT [TC] [47]: [INFO] Closing debug tool
2022-03-15 10:29:26 PDT [MT] [49]: [ERROR] TargetController failed to startup.
2022-03-15 10:29:26 PDT [MT] [50]: [INFO] Shutting down Bloom
2022-03-15 10:29:26 PDT [SH] [55]: [INFO] Shutting down SignalHandler
```
3. No big deal, I cycle power and restart bloom:
```bash
2022-03-15 10:30:58 PDT [MT] [2]: [INFO] Selected environment: "default"
[... 56 lines cut ...]
avrdude: stk500v2_jtag3_recv(): error in jtagmkII_recv()
Reading | #################                                  | 33% 0.24savrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: stk500v2_jtag3_recv(): error in jtagmkII_recv()
Reading | ################################################## | 100% 0.25s

avrdude: Device signature = 0x303030
avrdude: Expected signature for ATmega328P is 1E 95 0F
         Double check chip, or use -F to override this check.
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: stk500v2_jtag3_recv(): error in jtagmkII_recv()
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: jtag3_edbg_send(): Unexpected response 0x81, 0x11
avrdude: jtag3_edbg_recv(): Unexpected response 0x80
avrdude: jtag3_edbg_signoff(): unexpected response 0x81, 0x11
avrdude: jtag3_edbg_signoff(): unexpected response 0x01, 0x00

avrdude done.  Thank you.

make: *** [Makefile:147: flash] Error 1
```
7. No luck. I cycle power (on both devices) and same response.
8. The only way I am able to get back to step 1, is to use IPE to reprogram the device. IPE advises me that "The device might be in debugWIRE mode....Should debugWIRE be disabled...? To which I reply Yes. And I'm back to step 1.

lkoepsel, 2022-03-15:
I also just attempted the same steps, however I changed to: (*above it is false*)
```json
        "releasePostDebugSession": true
```
Same response both before and after cycling power.

lkoepsel, 2022-03-15:
I believe I found it: (*this was previously set false*)
```json
        "disableDebugWirePreDisconnect": true,
```
Once I set this to *true* and I set the above "releasePost..." back to *false*, I was able to exit *gdb*, which exited *bloom* and I was able to *make...make* again.

So my seemingly successful bloom.json is:
```json
{
  "environments": {
    "default": {
      "shutdownPostDebugSession": true,
      "debugTool": {
        "name": "snap",
        "releasePostDebugSession": false
      },

      "target": {
        "name": "atmega328p",
        "physicalInterface": "debug-wire",
        "disableDebugWirePreDisconnect": true,
         "updateDwenFuseBit": true
      },

      "debugServer": {
        "name": "avr-gdb-rsp",
        "ipAddress": "127.0.0.1",
        "port": "1442"
      }
    }
  },

  "insight": {
    "enabled": true
  }
}
```

I'll continue to test and post.

navnavnav, 2022-03-15:
Hi @lkoepsel 

Right, so the first issue you had was caused by that AVRDUDE timeout bug. AVRDUDE is not giving the SNAP enough time to perform the initial ISP connection attempt (which fails, initially, because the debugWire module is enabled). The fix for it has not yet been approved by the AVRDUDE maintainers (see the [PR](https://github.com/avrdudes/avrdude/pull/901)).

Once that fix is approved and you've installed the latest, that issue will no longer occur.

I can see why enabling `disableDebugWirePreDisconnect` fixes the issue for you - because it means Bloom will pull the target out of debugWire mode just before disconnecting. But I wouldn't recommend doing this as it will mean you'll have to cycle the target power every time you stop and start Bloom (even if you don't program the target). It's best to leave that disabled, and let the programming software pull the target out of debugWire mode. That way, you only have to cycle the power after you have programmed the target. But with that said, this is probably your best option until that issue in AVRDUDE is sorted.

I tested the changes with the SNAP and AVRDUDE - I have the same issue when I use the latest version of AVRDUDE. The issue disappears when I use my fixed version that's up for PR. If you're willing to build AVRDUDE yourself, you can just checkout my fork:

```
git clone git@github.com:navnavnav/avrdude.git;
git checkout bugfix/hid_read_timeout_too_low;
```

And then follow the build instructions at https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-Linux.
But I understand that may be a little too much work to bother yourself with.

lkoepsel, 2022-03-15:
Thanks!

Starting from the bottom...

Running ArchLinux is great until I need to do something like "build avrdude", then I run into quite a bit of LInux "stuff" which I don't want to have to learn. :)

Which means I'll leave the *disableDebugWirePreDisconnect* true and cycle power for now.

I'm going to do some testing with the AT328PB Mini Xplained...while I hate the name, I like the board.

lkoepsel, 2022-03-15:
The Xplained Mini works well! It seems to be a bit more solid, with the only issue is "cycle power" every exit of gdb/bloom to be able to re-enter gdb/bloom.

The power-cycling isn't a particularly bad issue, its fairly easy to solve with a small USB powered switch (for now). It will be nice to have the issue resolved.

I can also see your point as to not wanting to replace avrdude from a programming(upload) perspective. It would add some significant complexity to your efforts. I say that as I'm using Sublime Text and Make as my IDE and its pretty simple to compile/link/load via keystrokes. 

I'll provide a bit more detailed feedback as to 0.6.0 and Insight as I spend more time with it. I've been trying to come up with a significant lesson to demonstrate how to solve it using Insight.

navnavnav, 2022-03-16:
Excellent news on the Xplained Mini. I will now close this issue. If you have any other questions, feel free to start a discussion via the [Discussions](https://bloom.oscillate.io/discussions) page.

navnavnav, 2022-03-16:
Hi @lkoepsel 

Good news! It turns out that, for some debug tools (evaluation boards), we can cycle the target power without requiring any action from the user. I spent this afternoon implementing and testing this. It works really well with the Xplained Mini evaluation boards.

I've just shipped a very small feature release, v0.8.0, which provides target power management interfaces for the following debug tools:

- Xplained Mini
- Xplained Nano
- Xplained Pro
- Curiosity Nano

I'm not sure how many of those boards come with debugWire targets, but for those that do, Bloom will now automatically cycle target power after updating the DWEN fuse bit, if any of those debug tools are in use.

Please give it a try and let me know how it goes. I will reopen this issue so that you can provide feedback or ask any questions.

Download v0.8.0 from https://bloom.oscillate.io/download

Thanks

lkoepsel, 2022-03-16:
Wow, that’s interesting!

I’m tied up the rest of the day, I’ll test tomorrow and looking forward to it!

lkoepsel, 2022-03-16:
I was able to find some time to check out 0.8.0. And so far it isn't working for me:

I tried several different methods, however all of them ended with me having to physically cycle power on the 328PB Xplained.

Process:
1) Start Bloom then avr-gdb
2) Debug code, find something to change then quit avr-gdb, which shuts down bloom.
3) Modify code then make/make flash which uploads code via avrdude.
4) Attempt to start bloom and I get:
```bash
(.pyutil) lkoepsel@manjaroMBP Documents/AVR_C (main *) » bloom
2022-03-16 15:25:11 PDT [MT] [2]: [INFO] Selected environment: "default"
2022-03-16 15:25:11 PDT [TC] [6]: [INFO] Starting TargetController
2022-03-16 15:25:11 PDT [TC] [7]: [INFO] Connecting to debug tool
2022-03-16 15:25:11 PDT [TC] [9]: [INFO] Debug tool connected
2022-03-16 15:25:11 PDT [TC] [10]: [INFO] Debug tool name: Xplained Mini
2022-03-16 15:25:11 PDT [TC] [11]: [INFO] Debug tool serial: ATML2523052700008199
2022-03-16 15:25:11 PDT [TC] [13]: [WARNING] AVR8 debugWire interface selected - the DWEN fuse will need to be enabled
2022-03-16 15:25:11 PDT [TC] [14]: [INFO] Activating target
2022-03-16 15:25:12 PDT [TC] [31]: [WARNING] Failed to activate the debugWire physical interface - attempting to access target via the ISP interface, for DWEN fuse bit inspection.
2022-03-16 15:25:12 PDT [TC] [33]: [INFO] Initiating ISP interface
2022-03-16 15:25:12 PDT [TC] [34]: [WARNING] Updating the DWEN fuse bit is a potentially dangerous operation. Bloom is provided "AS IS", without warranty of any kind. You are using Bloom at your own risk. In no event shall the copyright owner or contributors be liable for any damage caused as a result of using Bloom. For more details, see the Bloom license at https://bloom.oscillate.io/license
2022-03-16 15:25:12 PDT [TC] [35]: [INFO] Reading target signature via ISP
2022-03-16 15:25:12 PDT [TC] [36]: [INFO] Target signature confirmed: 0x1e9516
2022-03-16 15:25:12 PDT [TC] [37]: [INFO] Current SPIEN fuse bit value confirmed
2022-03-16 15:25:12 PDT [TC] [39]: [INFO] Cycling target power
2022-03-16 15:25:13 PDT [TC] [43]: [INFO] Retrying debugWire physical interface activation
2022-03-16 15:25:13 PDT [TC] [45]: [ERROR] The TargetController encountered a fatal error. See below for errors:
2022-03-16 15:25:13 PDT [TC] [46]: [ERROR] Failed to activate the debugWire physical interface - check target connection. If the target was recently programmed via ISP, try cycling the target power. See https://bloom.oscillate.io/docs/debugging-avr-debugwire for more information.
2022-03-16 15:25:13 PDT [TC] [47]: [INFO] Shutting down TargetController
2022-03-16 15:25:13 PDT [TC] [48]: [INFO] Deactivating target
2022-03-16 15:25:13 PDT [TC] [49]: [INFO] Closing debug tool
2022-03-16 15:25:13 PDT [MT] [51]: [ERROR] TargetController failed to startup.
2022-03-16 15:25:13 PDT [MT] [52]: [INFO] Shutting down Bloom
2022-03-16 15:25:13 PDT [SH] [57]: [INFO] Shutting down SignalHandler
```
I have attempted this with variations of:
```json
        "cycleTargetPowerPostDwenUpdate": true and not there
        "disableDebugWirePreDisconnect": false and true,
```

My entire bloom.json is:
```json
{
  "environments": {
    "default": {
      "shutdownPostDebugSession": true,
      "debugTool": {
        "name": "xplained-mini",
        "releasePostDebugSession": false
      },

      "target": {
        "name": "atmega328pb",
        "physicalInterface": "debug-wire",
        "disableDebugWirePreDisconnect": false,
        "updateDwenFuseBit": true,
        "cycleTargetPowerPostDwenUpdate": true
      },

      "debugServer": {
        "name": "avr-gdb-rsp",
        "ipAddress": "127.0.0.1",
        "port": "1442"
      }
    }
  },

  "insight": {
    "enabled": true
  }
}
```

I have also confirmed I am running version 0.8.0
```bash
(.pyutil) lkoepsel@manjaroMBP Documents/AVR_C (main *) » bloom --version
Bloom v0.8.0
https://bloom.oscillate.io/
Nav Mohammed
```
Also confirmed via Bloom Insight -> About

What am I missing?

Thanks,
Lief

navnavnav, 2022-03-17:
Hmm, that's really odd. I cannot replicate the issue at all. I'm testing with an Xplained Mini Atmega168PB and the cycling of the target power works fine. I've tested after ISP programming and after a debug session with `disableDebugWirePreDisconnect` enabled - in all instances it works fine. Perhaps the Xplained Mini Atmega328PB doesn't support cycling target power. I don't have one of those to test with.

Do you have any other circuitry connected to the board? Have you made any modifications to the board? If not, then I may have to purchase an xmini with a 328pb and try to replicate with that.

navnavnav, 2022-03-17:
What's weird is, if the board didn't support cycling the power, then it should have reported a command failure, in response to the command to trigger the power cycling.

Hmm, then again, that command takes the form of just setting a parameter on the device (we set a certain param to 0x00 to turn power off, 0x01 to turn it on). So maybe that's why it didn't report the failure - the parameter was successfully set, but had no effect due to lack of support.

mraardvark, 2022-03-17:
I am fairly sure that these debugWIRE mini boards ALL support the same power toggle feature.

lkoepsel, 2022-03-17:
I believe you are correct, the debugWIRE mini boards would ALL support the same power toggle.

I believe one proof point would be IPE, as IPE is able to power-cycle the board (or at least it appears to be.)

Let's start with the same bloom.json and same steps (board types excepted).

Could you provide your bloom.json and the steps you are able to take?

Thanks
Lief

navnavnav, 2022-03-17:
Sure, here is my environment config:

```
"default": {
    "debugTool": {
      "name": "xplained-mini",
      "releasePostDebugSession": false
    },

    "target": {
      "name": "atmega168pb",
      "physicalInterface": "debug-wire",
      "disableDebugWirePreDisconnect": true,
      "updateDwenFuseBit": true,
      "cycleTargetPowerPostDwenUpdate": true
    },

    "shutdownPostDebugSession": true
  }
```

With that config, I can run Bloom without issue:

```
2022-03-17 14:52:09 GMT [MT] [2]: [INFO] Selected environment: "default"
2022-03-17 14:52:09 GMT [TC] [6]: [INFO] Starting TargetController
2022-03-17 14:52:09 GMT [TC] [7]: [INFO] Connecting to debug tool
2022-03-17 14:52:09 GMT [TC] [9]: [INFO] Debug tool connected
2022-03-17 14:52:09 GMT [TC] [10]: [INFO] Debug tool name: Xplained Mini
2022-03-17 14:52:09 GMT [TC] [11]: [INFO] Debug tool serial: ATML2222062700000201
2022-03-17 14:52:09 GMT [TC] [13]: [WARNING] AVR8 debugWire interface selected - the DWEN fuse will need to be enabled
2022-03-17 14:52:09 GMT [TC] [14]: [INFO] Activating target
2022-03-17 14:52:10 GMT [TC] [31]: [WARNING] Failed to activate the debugWire physical interface - attempting to access target via the ISP interface, for DWEN fuse bit inspection.
2022-03-17 14:52:10 GMT [TC] [33]: [INFO] Initiating ISP interface
2022-03-17 14:52:10 GMT [TC] [34]: [WARNING] Updating the DWEN fuse bit is a potentially dangerous operation. Bloom is provided "AS IS", without warranty of any kind. You are using Bloom at your own risk. In no event shall the copyright owner or contributors be liable for any damage caused as a result of using Bloom. For more details, see the Bloom license at https://bloom.oscillate.io/license
2022-03-17 14:52:10 GMT [TC] [35]: [INFO] Reading target signature via ISP
2022-03-17 14:52:10 GMT [TC] [36]: [INFO] Target signature confirmed: 0x1e9415
2022-03-17 14:52:10 GMT [TC] [37]: [INFO] Current SPIEN fuse bit value confirmed
2022-03-17 14:52:10 GMT [TC] [39]: [INFO] Cycling target power
2022-03-17 14:52:10 GMT [TC] [43]: [INFO] Retrying debugWire physical interface activation
2022-03-17 14:52:11 GMT [TC] [45]: [INFO] Target activated
2022-03-17 14:52:11 GMT [TC] [46]: [INFO] AVR8 target promoted to megaAVR target
2022-03-17 14:52:11 GMT [TC] [62]: [INFO] Target ID: 0x1e9415
2022-03-17 14:52:11 GMT [TC] [63]: [INFO] Target name: ATmega168PB
2022-03-17 14:52:11 GMT [MT] [66]: [INFO] Selected DebugServer: AVR GDB Remote Serial Protocol Debug Server
2022-03-17 14:52:11 GMT [DS] [67]: [INFO] Starting DebugServer
2022-03-17 14:52:11 GMT [DS] [72]: [INFO] GDB RSP address: 0.0.0.0
2022-03-17 14:52:11 GMT [DS] [73]: [INFO] GDB RSP port: 1442
2022-03-17 14:52:11 GMT [DS] [75]: [INFO] DebugServer ready
2022-03-17 14:52:11 GMT [DS] [76]: [INFO] Waiting for GDB RSP connection
2022-03-17 14:52:11 GMT [DS] [78]: [INFO] Waiting for GDB RSP connection
2022-03-17 14:52:11 GMT [MT] [79]: [INFO] Starting Insight
2022-03-17 14:52:11 GMT [MT] [84]: [INFO] Insight ready
^C2022-03-17 14:52:13 GMT [SH] [92]: [WARNING] Shutdown signal received
2022-03-17 14:52:13 GMT [SH] [93]: [INFO] Attempting clean shutdown
2022-03-17 14:52:13 GMT [MT] [97]: [INFO] Shutting down Bloom
2022-03-17 14:52:13 GMT [MT] [98]: [INFO] Shutting down Insight
2022-03-17 14:52:13 GMT [DS] [102]: [INFO] Shutting down DebugServer
2022-03-17 14:52:13 GMT [TC] [108]: [INFO] Shutting down TargetController
2022-03-17 14:52:13 GMT [TC] [109]: [INFO] Deactivating target
2022-03-17 14:52:13 GMT [TC] [110]: [INFO] Closing debug tool
2022-03-17 14:52:13 GMT [SH] [116]: [INFO] Shutting down SignalHandler
```

Then, I program the target via avrdude:
```
avrdude: stk500v2_command(): command failed
avrdude: Target prepared for ISP, signed off.
avrdude: Now retrying without power-cycling the target.
avrdude: AVR device initialized and ready to accept instructions
[... 47 lines cut ...]
2022-03-17 14:55:20 GMT [MT] [66]: [INFO] Selected DebugServer: AVR GDB Remote Serial Protocol Debug Server
2022-03-17 14:55:20 GMT [DS] [67]: [INFO] Starting DebugServer
2022-03-17 14:55:20 GMT [DS] [72]: [INFO] GDB RSP address: 0.0.0.0
2022-03-17 14:55:20 GMT [DS] [73]: [INFO] GDB RSP port: 1442
2022-03-17 14:55:20 GMT [DS] [75]: [INFO] DebugServer ready
2022-03-17 14:55:20 GMT [DS] [76]: [INFO] Waiting for GDB RSP connection
2022-03-17 14:55:20 GMT [DS] [78]: [INFO] Waiting for GDB RSP connection
2022-03-17 14:55:20 GMT [MT] [79]: [INFO] Starting Insight
2022-03-17 14:55:20 GMT [MT] [84]: [INFO] Insight ready
^C2022-03-17 14:55:25 GMT [SH] [92]: [WARNING] Shutdown signal received
2022-03-17 14:55:25 GMT [SH] [93]: [INFO] Attempting clean shutdown
2022-03-17 14:55:25 GMT [MT] [97]: [INFO] Shutting down Bloom
2022-03-17 14:55:25 GMT [MT] [98]: [INFO] Shutting down Insight
2022-03-17 14:55:25 GMT [DS] [102]: [INFO] Shutting down DebugServer
2022-03-17 14:55:25 GMT [TC] [108]: [INFO] Shutting down TargetController
2022-03-17 14:55:25 GMT [TC] [109]: [INFO] Deactivating target
2022-03-17 14:55:25 GMT [TC] [110]: [INFO] Closing debug tool
2022-03-17 14:55:25 GMT [SH] [116]: [INFO] Shutting down SignalHandler
```

I can do this any number of times and it never fails.

@lkoepsel if you have no luck with that config, I will try to create a new build for you, with an increased delay in the power cycle routine. Currently, we hold the power off for 250 milliseconds, which I assumed would be enough. But maybe it's not.

Have you made any modifications to the board? Have you added any capacitors anywhere on the board?

lkoepsel, 2022-03-17:
And no capacitors on board.
I have an untouched 328PB Xplained Mini which I will test next if the delay doesn't work...I'd rather leave it untouched if possible.

An image of my setup is this:
https://wellys.com/posts/images/328PB_testbed.png

navnavnav, 2022-03-17:
Sorry @lkoepsel, I've just realised that build won't work unless it's in a certain location.

I've prepared a debian package for you: https://bloom-public-bucket.s3.eu-west-1.amazonaws.com/misc/Bloom-0.8.1-Linux.deb

Download and install that:

```
sudo apt-get remove bloom;
sudo apt install path/to/bloom.deb;
```

lkoepsel, 2022-03-17:
I have ArchLinux, so I need to make this a pacman package...not a big deal, just wanted you to know.

I made the change, confirmed version was 8.1 and to no avail. There is a definite delay at "Cycling Target Power", however same issue "TargetController encountered a fatal error..."

And physically cycling power fixes the problem.
LIef

navnavnav, 2022-03-17:
That's so weird. Can you connect an LED to the VCC and GND pins at the bottom right of the board:

[image]

And then run Bloom again. The LED should turn off for 5 seconds and then back on. Let me know how that goes

Thanks

lkoepsel, 2022-03-17:
Let add this behavior as well:
1) If I attempt to program via IPE, the messages come up "Failed to connect to device using ISP...Should debugWIRE be disabled and ISP enabled". When I click on Yes, the LED does not go out, however, the board gets programmed.
2) And I still have to cycle power to get Bloom to work.

navnavnav, 2022-03-17:
Yikes! Now I'm really lost. I'm just going to cave and buy one of those boards. It should be here sometime next week. I'll try to replicate the issue on my side and see if anything can be done.

> When I click on Yes, the LED does not go out, however, the board gets programmed.

Yeah that makes sense, as IPE will pull the target out of debugWire mode. But you shouldn't have to cycle the power if we're already doing it. It's really weird.

I'll let update this ticket as soon as I know more. Thank you for your help with this, Lief.

Just as a last-ditch effort, have you tried removing the board from the two breadboards to which it was connected in the image you shared?

lkoepsel, 2022-03-17:
Not yet, I will begin to play with it to see if I can find any issues.

Depopulating the board would be one of items to do along with testing if other VCC on the board points don't go dark. 

I'll be out for few hours and will respond much later today.

navnavnav, 2022-03-17:
Is this resistor connected to the AREF pin?
[image]

If it is, it could be the culprit. I'd start with removing everything on the AREF pin.

navnavnav, 2022-03-21:
Hi @lkoepsel 

So my Xplained Mini 328PB arrived today, and I was able to replicate the issue with it. After looking into this, I've found the cause to be a lack of time allowance for the target to power back up, during the power cycle.

Initially, during the power cycling routine, we would disable the power, wait for 250 ms, then enable the power and immediately attempt another debugWire connection. We were not allowing the target any time to power-up, after enabling the power and attempting another connection. This wasn't an issue for the Xplained Mini 168PB, but clearly was for the 328PB board.

You can see this for yourself, by simply running Bloom again, after it fails to connect, **without manually cycling the target power**, and you'll see that it works. So if you do the following:

- Run Bloom for the first time, with the `disableDebugWirePreDisconnect` option enabled - it should work here
- Close Bloom so that it disables the debugWire module
- Run it again - it should fail to connect here, even though it successfully cycles the target power
- Then run it again (without manually cycling the target power). **It should work here** because the target has had enough time to power-up.

So yeah, that was all it was in the end.

I have prepared a fix for this in version 0.8.2, which is now available for you to download at: https://bloom.oscillate.io/download

Let me know how that goes.
Thanks!

lkoepsel, 2022-03-21:
WORKS!!! 
And its great, the roundtrip now is pretty easy!

Thanks and great work!
Lief

navnavnav, 2022-03-21:
Great news! Thanks for confirming @lkoepsel 

I will now close this issue. Feel free to open another if you have any other issues.

closed 2022-03-21 by navnavnav