bloombloombloom/Bloom #28: Distribute Bloom via Pacman package, for Arch Linux users

navnavnav, 2022-04-04:
I've noticed a number of Bloom users on Arch Linux. In order to install Bloom, some of them convert the Debian package into another format. It would be nice if we could provide an Arch Linux compatible package for them.

navnavnav, 2022-04-12:
@maximevince Thanks for this!

Is that a "build from source" package? I'm not very familiar with package management on Arch Linux. The Debian and RPM packages are binary packages. They also include shared objects of some third-party libraries for various reasons (licensing, lack of availability in some package management systems, etc).

Hmm, I'm not sure if that PHP 7 dependency is quite right. PHP 8 is required to build Bloom (for the TDF validation scripts). Those validation scripts use PHP features that are only available in PHP 8.

Honestly, I wouldn't bother yourself with a "build from source" package. It would be a lot of pain. IMO it's a lot easier to ship pre-built binary packages.

BTW I hope to get an Arch Linux package sorted after the [TargetController refactor work](https://github.com/navnavnav/Bloom/issues/25). I'm not sure when that will be, but will keep this ticket updated 👍🏽

maximevince, 2022-04-14:
Hey @navnavnav,

With pleasure.
On Arch, this is the way users can add support for a package, when it's not in the official binary repo's yet.
It's really a single command to install this, for every Arch user now: e.g. `yay -S bloom-git`.
The build from source procedure is what I've described in the "PKGBUILD" file (you can see it here: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=bloom-git).
It's building and running fine using that PKGBUILD.

Arch is a rolling release with mostly "bleeding-edge" packages, so PHP is version 8.1.4 right now, for all Arch users. So that's not a problem either.

That being said, I agree that an official binary build would be even better!

navnavnav, 2022-05-07:
Version 0.9.0 has just been released. This includes support for Pacman packages. See the Bloom download page (https://bloom.oscillate.io/download) to download the Pacman package. Closing this now.

closed 2022-05-07 by navnavnav