bloombloombloom/Bloom #31: Support for the MPLAB ICE 4

navnavnav, 2022-04-10:
I hope to purchase one of these at some point (but not sure when). It would be nice if Bloom could support it.

I will keep this ticket updated with any developments.

xedbg, 2022-04-11:
FYI: ICE4 is not "EDBG-based" so all bets are off with regard to the protocol, so expect to have to do a fair amount of work to get it in.

navnavnav, 2022-04-11:
Thanks for letting me know, @xedbg 

Do you know when/if Microchip plan on publishing any documents RE the protocols employed by the ICE 4? I won't be able to support it without some documentation. Anything similar to the "EDBG-based Tools Protocols" document would be excellent.

xedbg, 2022-04-11:
I can ask around... (but PICkit4 and Snap are part of that protocol-group and those are not public [yet])

xedbg, 2022-05-30:
The short answer is: ICE 4 actually _does_ use the same EDBG-based stack for supporting AVR devices over a USB connection, so the same code supporting PICkit4 and Snap should in theory work for ICE4
The longer answer is: its complicated - this may be a short/medium-term solution, and at some point the interface may be changed.  There are currently no concrete plans to publish that protocol spec, but that too may change in the longer-term.  These units are also well above the price bracket for "giveaways", so for your purposes - any hardware investment which expects an EDBG-like interface has a fair amount of risk attached, I'm afraid.

navnavnav, 2022-05-30:
Thanks for letting me know @xedbg.

I will hold off on this for now, seeing as no one has requested it.

closed 2023-12-31 by navnavnav