bloombloombloom/Bloom #32: JTAGICE3 supported?

maximevince, 2022-04-11:
Hi,

I am not trying to connect to an ATxmega128D3 target, using JTAGICE3 from Atmel.
Is this a supported debugger? I could find the vid:pid definition in the `99-bloom.rules` file, but no references to it in the code.

Thanks,
Maxime

xedbg, 2022-04-11:
See discussion: https://github.com/navnavnav/Bloom/discussions/30

maximevince, 2022-04-11:
@xedbg Thanks. Going on that information, I just went out on a limb an guessed the 3.x firmware of the JTAGICE3 could possible implemented the EDBG protocol in the same way as the Atmel-ICE.

I tried this:
```patch
diff --git a/src/DebugToolDrivers/Microchip/AtmelICE/AtmelIce.hpp b/src/DebugToolDrivers/Microchip/AtmelICE/AtmelIce.hpp
index cd43e03..7a7ee16 100644
--- a/src/DebugToolDrivers/Microchip/AtmelICE/AtmelIce.hpp
+++ b/src/DebugToolDrivers/Microchip/AtmelICE/AtmelIce.hpp
@@ -43,7 +43,8 @@ namespace Bloom::DebugToolDrivers
     {
     public:
         static const std::uint16_t USB_VENDOR_ID = 1003;
-        static const std::uint16_t USB_PRODUCT_ID = 8513;
+        static const std::uint16_t USB_PRODUCT_ID = 8512;
 
         AtmelIce(): UsbDevice(AtmelIce::USB_VENDOR_ID, AtmelIce::USB_PRODUCT_ID) {}
```
And it just connected!

[image]

I am willing to create a patch for this, but can you let me know if you'd rather:
* Have the `AtmelIce.cpp` code try both known USB vid:pid pairs?
* Or create an additional `AtmelJtagIce3.cpp` class?

navnavnav, 2022-04-11:
Thanks Maxime, I appreciate you taking the time to try it out.

I feel really unconformable with shipping changes that I've not been able to test myself, so I'd only wish to ship them once I've gotten hold of a JTAGICE3 and tested the changes.

However, there is no reason why I can't create an unofficial release for you and @septatrix, with the basic changes. At least the two of you will then have something to work with.

I'm surprised (and glad) you were able to build Bloom, @maximevince! The CMake configuration isn't in the best of shape and dependencies aren't very will documented ([something I plan to address soon](https://github.com/navnavnav/Bloom/issues/26)). But I'm glad you managed to do it!

I will have that unofficial release ready for you either later today, or later tomorrow.

I will close this issue and move this discussion to the GitHub discussion https://github.com/navnavnav/Bloom/discussions/30. Will post any updates on there.

Thanks!