bloombloombloom/Bloom #33: Redditor trying to debug an AVR bootloader. Bloom having packet issues.

CathyGarrett, 2022-04-18:
I'm trying to develop this firmware as a bootloader, but it's executing off into the weeds and I'm trying to catch it in the act.

```
$ ddd --debugger "avr-gdb -ex 'file cmake-build-debug-avr/firmware.elf' -ex 'target remote :gdbremote'"

GNU DDD 3.3.12 (x86_64-unknown-linux-gnu), by Dorothea LReading symbols from cmake-build-debug-avr/firmware.elf...
Remote debugging using :gdbremote
0x00000000 in __vectors ()
```

I add a breakpoint at the top of main to skip past all the startup noise.

```
(gdb) break /home/garrett/build/git/firmware/src/firmware.cpp:46
Breakpoint 1 at 0x6ca: file /home/garrett/build/git/firmware/src/firmware.cpp, line 46.
(gdb) cont
Continuing.

Breakpoint 1, main () at /home/garrett/build/git/firmware/src/firmware.cpp:48

```
Now, I set a breakpoint down in the body of main(). If I set the breakpoint before the problem instruction, it'll continue straight to that breakpoint. If I set it after the problem instruction, that instruction will execute off into the weeds and cause a reset, which will ultimately be caught at the breakpoint at the top of main() again.

```
(gdb) break *0x0000072c
Breakpoint 2 at 0x72c: file /home/garrett/build/git/firmware/src/firmware.cpp, line 79.
(gdb) cont
Continuing.

Breakpoint 2, 0x0000072c in main () at /home/garrett/build/git/firmware/src/firmware.cpp:79
```

Okay. It's gotta be deeper. Let's move the breakpoint one instruction later.

```
(gdb) delete 2
(gdb) break *0x0000072e
Breakpoint 3 at 0x72e: file /home/garrett/build/git/firmware/src/firmware.cpp, line 79.
(gdb) cont
Continuing.

Breakpoint 1, main () at /home/garrett/build/git/firmware/src/firmware.cpp:48
```

Okay. I think I found my culprit, but let's reset the breakpoint higher and continue to see what happens.

```
(gdb) delete 3
(gdb) break *0x0000072c
Breakpoint 4 at 0x72c: file /home/garrett/build/git/firmware/src/firmware.cpp, line 79.
(gdb) cont
Continuing.
Continuing.
Continuing.
Continuing.
Continuing.
Ignoring packet error, continuing...

Breakpoint 1, main () at /home/garrett/build/git/firmware/src/firmware.cpp:48
```

Okay. That's not good. What's with the repeated "Continuing." messages? And there's the packet error. Let's see if it eventually recovers and gets back to the top of main().

```
(gdb) cont
Continuing.
Ignoring packet error, continuing...

Breakpoint 1, main () at /home/garrett/build/git/firmware/src/firmware.cpp:48

[... 324 lines cut ...]
      "debugTool": {
        "name": "atmel-ice",
        "releasePostDebugSession": true
      },

      "target": {
        "name": "at90can128",
        "physicalInterface": "jtag"
      },

      "debugServer": {
        "name": "avr-gdb-rsp",
        "ipAddress": "127.0.0.1",
        "port": "2159"
      }
    }
  },

  "insight": {
    "enabled": true
  }
}
```

I use port 2159, because it's actually listed in my /etc/services file, labelled gdbremote, so that's why I use that in my ddd invocation. Would be really convenient if bloom did local port name resolution rather than demanding numerical port numbers.

navnavnav, 2022-04-19:
Thanks for reporting this, @CathyGarrett 

It's really odd - I don't understand why GDB is complaining about the response to the `c` (continue) command packet. Simply put, there is no response to that packet. Bloom only acknowledges receipt of the packet. No actual response packet is sent to GDB in response to the `c` command packet, until the target halts. When the target halts, Bloom will send a 'stop reply' packet.

I think the full debug output would be really helpful here. Can you please send over the full debug output? Just put it in a text file and upload it with your comment. You can upload ASCII text files and include them in your comment by either dragging and dropping the file in the comment text input, or just click on the attachment bar:

[image]

Thanks

navnavnav, 2022-04-19:
Also, could you please try to replicate the issue with Insight disabled?

That error you're seeing ("Invalid target state - target must be stopped before pin states can be...") is a result of Insight attempting to pull the target pin states whilst the target is running. In most cases, it should be nothing to worry about, but the fact that you're having GDB issues when this error occurs is a concern for me. If you disable Insight (by setting the `enabled` flag to `false`, in the `insight` object, in your bloom.json), then that error should no longer occur. Please try that and let me know if you still face the same GDB issues. And please attach Bloom's full debug log output here.

Thanks!

navnavnav, 2022-12-10:
Hi @CathyGarrett,

Seeing as I haven't heard back from you in some time, I'm going to close this issue.

I suspect that those "Ignoring packet error, continuing..." messages were the result of Bloom taking too long to respond to GDB, and GDB timing out.

There have been a number of updates to Bloom since this issue was created. I believe some of those updates should have fixed this (as they ensure that Bloom responds to GDB in a timely manner). If you haven't already, I suggest upgrading Bloom and check if the issue persists. If it does, feel free to reopen this issue with a full debug log.

Thanks!

closed 2022-12-10 by navnavnav