bloombloombloom/Bloom #39: Provide machine readable target register information to GDB frontends (IDEs)

navnavnav, 2022-05-13:
Some IDEs (like CLion) allow users to provide an SVD file, from which the IDE will extract target information. To make things easier, it would be nice if Bloom could provide these IDEs with the information they require. 

Work began on this in version 0.9.0, where GDB monitor commands were added to allow IDEs to extract information from Bloom. In response to these commands, Bloom will output the requested information in JSON format. The commands can be issued at any point during a debug session with Bloom.

The `version machine` command:
```
(gdb) monitor version machine
{
    "components": {
        "major": 0,
        "minor": 9,
        "patch": 0
    },
    "version": "0.9.0"
}
``` 

The `target-info machine` command:
```
(gdb) monitor target-info machine
{
    "target": {
        "id": "0x1e950f",
        "memoryDescriptors": [
            {
                "addressRange": {
                    "endAddress": 32767,
                    "startAddress": 0
                },
                "name": "Flash",
                "size": 32768
            },
            {
                "addressRange": {
                    "endAddress": 2303,
                    "startAddress": 256
                },
                "name": "RAM",
                "size": 2048
            },
            {
                "addressRange": {
                    "endAddress": 1023,
                    "startAddress": 0
                },
                "name": "EEPROM",
                "size": 1024
            }
        ],
        "name": "ATmega328P"
    }
}
``` 

I now want to extend these commands to include information on target registers. I'm not sure if this will be a separate command like `register-info machine`, or just included in the response to the `target-info machine` command. The register information will include names, descriptions, plain addresses, sizes, addresses with relevant offsets applied for access via avr-gdb, readable and writable flags, etc.

This would be a nice way to provide IDEs with the information they need to provide a better experience for Bloom users. I will also document each command on the Bloom website, at some point, for IDE developers to use as a reference.

Not sure when I'll get onto this. It will likely be looked at after [support for programming](https://github.com/navnavnav/Bloom/issues/23) is completed.

navnavnav, 2022-05-13:
BTW, I'm open to suggestions on this. If any IDE developers wish to add anything, or suggest an alternative method of providing information to IDEs, feel free to leave a comment here. Would be great to hear from any IDE developers on how Bloom can assist them.

septatrix, 2022-05-15:
There does not seem to be any widespread adoption of such a feature in GDB servers and as such I do not think any IDE will really implement this just for bloom. Most of them just allow the user to configure an svd file as e.g. the memory size but also other stuff nowadays often depends not only on the chip but also variants and external SPI which is rather uncommon in AVR land. So I guess this would be mostly of interest for plug-ins/extensions integrating bloom in some IDE though I know of any example for that.

So IDEs will likely simply keep their svd way and CLI users can already use something like [svd-tools](https://github.com/1udo6arre/svd-tools). Both obviously require SVDs but that is probably best kept to a separate tool.

navnavnav, 2022-05-15:
Thanks for your thoughts on this @septatrix. The idea behind this is to relieve some of the pain from users having to generate SVD files. They may not be aware of the tools available to convert Microchips ATDF files. In fact, they may not be aware of ATDF files at all.

As it stands, users have to find and download the ATDF files, find, install or build from source a tool that can convert the file to SVD, apply the GDB offset or change each address in the SVD file and then point their IDE to it. That's a lot of work for someone who just wants to debug their code.

But perhaps you're right. I will put this ticket into 'in consideration' and wait for IDE developers to request it, before I begin work on it.

navnavnav, 2022-06-05:
Including register descriptors in the output of the `target-info machine` was easy enough to do, so I just did it. If IDE developers want more data, they'll have to ask for it.

closed 2022-06-05 by navnavnav