bloombloombloom/Bloom #43: Compiling on a Raspberry PI 4

oltdaniel, 2022-06-24:
For a project I need to run Bloom on a Raspberry PI. Or to be precise, I need to run the Bloom GDB Server on a Raspberry PI as I'm nut using the Insight Window. This still requires me, to fully install QT6 (which is a challenge in itself) and compile it for ARM64. But in the end it worked like a charm so I want to share it for anybody else who tries to do this and save them a few hours.

### Prepare

First of all, we some custom online repositories in order to acquire the necessary dependencies,

#### Basic tools

```
# Some build tools
sudo apt install cmake build-essential git wget
# Some other required libraries
sudo apt install libusb-1.0-0-dev libhidapi-dev
```

#### QT 6

```
# Import the SID debian package repository (no clue what it is, but it contains qt6 for arm64 and is official)
echo "deb http://ftp.de.debian.org/debian sid main" | sudo tee /etc/apt/sources.list.d/sid.list
# Now install dozen of qt6 packages
sudo apt update
sudo apt install qt6-base-dev qt6-tools-dev libqt6uitools6 libqt6svg6-dev libvulkan-dev libxkbcommon-x11-dev
```

> **NOTE**: Somebody more familiar with this can probably even simplify these dependencies for QT6. But in the end this worked for me.

#### QT 5

> Wait what? Yes we need some QT5 tool `/usr/bin/qt5/rcc` if I remember correctly

```
# Install qt5
sudo apt install qtbase5-dev
```

#### PHP 8

```
# Load the gpp key
sudo wget -qO /etc/apt/trusted.gpg.d/php.gpg https://packages.sury.org/php/apt.gpg
# Register the package repository
echo "deb https://packages.sury.org/php/ $(lsb_release -sc) main" | sudo tee /etc/apt/sources.list.d/php.list
# Install the PHP dependencies
sudo apt install php8.0-cli php8.0-xml
```

### Compile

```
git clone https://github.com/navnavnav/Bloom
cd Bloom
mkdir build/cmake-build-debug
cd build/cmake-build-debug
cmake ../..
cd ../..
cmake --build ./build/cmake-build-debug --target Bloom
```

@navnavnav It would be really nice if we could disconnect the Insight and GDB Logic from another in the build process so the QT dependencies aren't always required. I would be willing to contribute as this is something I may need anyway, depending on the choice of my boss in the next weeks.

> And thanks for Bloom. Else we couldn't use our current and old hardware to modernize the software and debugging part of the project.

navnavnav, 2022-06-26:
> Yes we need some QT5 tool

Why couldn't you use the Resource Compiler (rcc) from your Qt6 installation? You shouldn't need to install Qt5 for rcc.

> It would be really nice if we could disconnect the Insight and GDB Logic from another in the build process so the QT dependencies aren't always required. I would be willing to contribute as this is something I may need anyway, depending on the choice of my boss in the next weeks.

This has already been discussed with your associate. The work required is substantial and is simply not justified at this point in time (you and your associate are the only ones who have requested it - to my knowledge, there has been no other demand for it). 

Furthermore, removing Insight at compile time will not completely remove the Qt dependency - we use Qt in some other areas like processing JSON project configuration files (bloom.json), processing XML target description files, string manipulation (only where `std::string` falls short), etc. This means you will still have to install Qt6 to compile Bloom (but you'd only have to install it on the one Raspberry Pi where you build Bloom. After that, you can distribute Qt's shared objects alongside Bloom's binary, to your other Pis, without having to install Qt or other graphics related dependencies). So removing Insight will only remove the graphical dependencies, but not all Qt dependencies.

OK, here is what I propose:

I will do this for you under the following conditions:

- This configuration will not be used in any official Bloom packages. Users who wish to use this configuration will have to manually compile Bloom, as you have done.
- This configuration will not be tested against any subsequent changes to Bloom. If any changes result in issues with the configuration, users will have to discover and report these before they are looked at. In other words, they will effectively be using an untested version of Bloom.
- This configuration will only remove the Qt **GUI** dependencies - Qt will remain a dependency (for reasons mentioned above)
- Work will only begin on this once either of you have confirmed that you actually intend to use it. It is not fair to expect me to do weeks/months of work, for something that you \*\*might\*\* use. I need to be sure that you're actually going to use Bloom before I invest the time.

These conditions will reduce the strain on me and allow you to have something you can work with.

oltdaniel, 2022-06-28:
> Why couldn't you use the Resource Compiler (rcc) from your Qt6 installation? You shouldn't need to install Qt5 for rcc.

Turns out, `qtchooser` in this case at least, can only list `qt4` to `qt5`. So the `rcc` which is an alias for `qtchooser` falls back to `qt5`. So as `qtchooser` in this case doesn't seem to support `qt6`, the easiest workaround is to install the `qt5` version as well. Stupid, but debian won't let me in this case.

> These conditions will reduce the strain on me and allow you to have something you can work with.

Ok as this seems to be a large workaround, I think I'll try to compile everything again on a clean setup to verify if everything works without having the desktop installed.

kristianlm, 2022-11-01:
Hi,

For what it's worth I've been trying to package Bloom for GNU Guix. They aren't fan of distributing pre-built blobs so I have to remove those. This, and a few other issues, makes this incredeble piece of software 🤩 very hard to build for this distro unfortunately (so I've given up for now).

I'm in @oltdaniel's shoes in that I don't need a GUI, even though it's very polished. The other open-source [options](https://github.com/dcwbrown/dwire-debug) [options](https://github.com/dwtk/dwtk) [don't](https://github.com/rafaelmartins/dwire-gdb) [seem](https://github.com/mvirkkunen/dwprog) [to](https://github.com/ccrause/debugwire-gdb-bridge) [be](https://github.com/rafaelmartins/dwire-gdb) maintained. I would _love_ to have just `bloom-cli` that starts a gdb server.

I'm saying this in the hope that it will inspire you, @navnavnav, to consider separating this into two projects over time. A non-GUI part could do all the heavy lifting (and hopefully becomes easier to maintain) and a separate GUI could sit on top of it. 

Apart from that, thank you for enabling me to debug my AVR chips!

navnavnav, 2022-11-01:
Hi @kristianlm,

Thanks for raising this.

Since the discussion to disable Insight at compile time was started, a number of others have request something similar for their own specific reasons. So perhaps it's worth reconsidering this.

I have created ticket https://github.com/bloombloombloom/Bloom/issues/57. I can't make any promises, but I'll see what I can do. If I do it, it's unlikely it will be released this year. Early 2023 is more likely ATM, though that may change.

I'll post any further updates on that ticket.

I'm going to close this issue now as the new ticket will take its place.

closed 2022-11-01 by navnavnav