bloombloombloom/Bloom #44: libGL error: MESA-LOADER...

lkoepsel, 2022-06-29:
First, see my note in Discussions as to success!
Second, I did have this error occur, however, it didn't seem to affect the loading of programs. I haven't fully tested 0.10 and Insight, so I might be able to report more by the end of the week.

Here is the console where the error occurred:
```
2022-06-29 11:30:18 PDT [TC] [34]: [INFO] Reading target signature via ISP
2022-06-29 11:30:18 PDT [TC] [35]: [INFO] Target signature confirmed: 0x1e950f
2022-06-29 11:30:18 PDT [TC] [36]: [INFO] Current SPIEN fuse bit value confirmed
2022-06-29 11:30:18 PDT [TC] [37]: [INFO] Cleared lock bits confirmed
2022-06-29 11:30:18 PDT [TC] [38]: [WARNING] Programming DWEN fuse bit
2022-06-29 11:30:19 PDT [TC] [39]: [INFO] DWEN fuse bit successfully updated
2022-06-29 11:30:19 PDT [TC] [40]: [INFO] Retrying debugWire physical interface activation
2022-06-29 11:30:19 PDT [TC] [41]: [INFO] Target activated
2022-06-29 11:30:19 PDT [TC] [42]: [INFO] AVR8 target promoted to megaAVR target
2022-06-29 11:30:20 PDT [TC] [58]: [INFO] Target ID: 0x1e950f
2022-06-29 11:30:20 PDT [TC] [59]: [INFO] Target name: ATmega328P
2022-06-29 11:30:20 PDT [DS] [62]: [INFO] Starting DebugServer
2022-06-29 11:30:20 PDT [DS] [63]: [INFO] Selected DebugServer: AVR GDB Remote Serial Protocol Debug Server
2022-06-29 11:30:20 PDT [DS] [64]: [INFO] GDB RSP address: 127.0.0.1
2022-06-29 11:30:20 PDT [DS] [65]: [INFO] GDB RSP port: 1442
2022-06-29 11:30:20 PDT [DS] [69]: [INFO] DebugServer ready
2022-06-29 11:30:20 PDT [DS] [70]: [INFO] Waiting for GDB RSP connection
2022-06-29 11:30:20 PDT [MT] [71]: [INFO] Starting Insight
libGL error: MESA-LOADER: failed to open crocus: /opt/bloom/bin/lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/crocus_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open crocus: /opt/bloom/bin/lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/crocus_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open swrast: /opt/bloom/bin/lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
2022-06-29 11:30:20 PDT [MT] [75]: [INFO] Insight ready
2022-06-29 11:31:03 PDT [DS] [79]: [INFO] Accepted GDP RSP connection from 127.0.0.1
2022-06-29 11:31:12 PDT [TC] [160]: [WARNING] Programming mode enabled
2022-06-29 11:31:16 PDT [TC] [185]: [INFO] Programming mode disabled
2022-06-29 11:34:20 PDT [TC] [290]: [WARNING] Programming mode enabled
2022-06-29 11:34:23 PDT [TC] [315]: [INFO] Programming mode disabled
```

The last four lines were successful connections with the target.

Your thoughts?

Thanks,
Lief

navnavnav, 2022-06-30:
Hi Lief,

Thanks for reporting this. I believe I've seen this error before. I suspect it's down to the libstdc++ shared object that I ship with Bloom's binary.

I plan to look into it further when I get time, but to my knowledge, it doesn't cause any issues so it's not a priority. If you do stumble across an issue that you suspect could be caused by this, let me know and I'll make it a priority. 

I will update this ticket when I know more, but in the meantime, feel free to just ignore that error message.

Thanks!

navnavnav, 2023-11-02:
I believe this was fixed in v1.0.0. We no longer distribute a libstdc++ shared object in the Pacman package.

Will close this now, but feel free to reopen if it's still a problem.

Thanks,