bloombloombloom/Bloom #56: Missing "#include <optional>" in ProjectConfig.hpp?

shuyuan-liu, 2022-10-09:
I was compiling Bloom from source on Arch Linux with GCC 12.2.0 and GNU libc 2.36. GCC gave errors saying ProjectConfig.hpp used `std::optional` without including `<optional>`:

```
[  4%] Building CXX object CMakeFiles/Bloom.dir/src/Logger/Logger.cpp.o
In file included from /home/shuyuan/Build/Bloom/src/Logger/Logger.hpp:8,
                 from /home/shuyuan/Build/Bloom/src/Logger/Logger.cpp:1:
/home/shuyuan/Build/Bloom/./src/ProjectConfig.hpp:61:14: error: ‘optional’ in namespace ‘std’ does not name a template type
   61 |         std::optional<std::string> variantName;
      |              ^~~~~~~~
/home/shuyuan/Build/Bloom/./src/ProjectConfig.hpp:7:1: note: ‘std::optional’ is defined in header ‘<optional>’; did you forget to ‘#include <optional>’?
    6 | #include <yaml-cpp/yaml.h>
  +++ |+#include <optional>
    7 | 
/home/shuyuan/Build/Bloom/./src/ProjectConfig.hpp:195:14: error: ‘optional’ in namespace ‘std’ does not name a template type
  195 |         std::optional<DebugServerConfig> debugServerConfig;
      |              ^~~~~~~~
/home/shuyuan/Build/Bloom/./src/ProjectConfig.hpp:195:9: note: ‘std::optional’ is defined in header ‘<optional>’; did you forget to ‘#include <optional>’?
  195 |         std::optional<DebugServerConfig> debugServerConfig;
      |         ^~~
/home/shuyuan/Build/Bloom/./src/ProjectConfig.hpp:200:14: error: ‘optional’ in namespace ‘std’ does not name a template type
  200 |         std::optional<InsightConfig> insightConfig;
      |              ^~~~~~~~
/home/shuyuan/Build/Bloom/./src/ProjectConfig.hpp:200:9: note: ‘std::optional’ is defined in header ‘<optional>’; did you forget to ‘#include <optional>’?
  200 |         std::optional<InsightConfig> insightConfig;
      |         ^~~
/home/shuyuan/Build/Bloom/./src/ProjectConfig.hpp:225:14: error: ‘optional’ in namespace ‘std’ does not name a template type
  225 |         std::optional<DebugServerConfig> debugServerConfig;
      |              ^~~~~~~~
/home/shuyuan/Build/Bloom/./src/ProjectConfig.hpp:225:9: note: ‘std::optional’ is defined in header ‘<optional>’; did you forget to ‘#include <optional>’?
  225 |         std::optional<DebugServerConfig> debugServerConfig;
      |         ^~~
/home/shuyuan/Build/Bloom/./src/ProjectConfig.hpp:231:14: error: ‘optional’ in namespace ‘std’ does not name a template type
  231 |         std::optional<InsightConfig> insightConfig;
      |              ^~~~~~~~
/home/shuyuan/Build/Bloom/./src/ProjectConfig.hpp:231:9: note: ‘std::optional’ is defined in header ‘<optional>’; did you forget to ‘#include <optional>’?
  231 |         std::optional<InsightConfig> insightConfig;
      |         ^~~
```

Clang 14.0.6 gave similar errors. I grep'd the src directory for `optional` and at a glance it seems that most other files that use it have included the header, but ProjectConfig.hpp hasn't.

navnavnav, 2022-10-09:
Hi @shuyuan-liu 

Thanks for reporting this.

Yes, the `<optional>` header should have been included there. I will fix this in the next release. In the meantime, you can just include it yourself to get the compilation to work.

Oddly, I don't get this error when compiling with G++10. Are you using CMake to trigger the build?

Also, may I ask why you're building from source? I distribute an Arch package via the Download page on the Bloom website: https://bloom.oscillate.io/download. Have you been having issues with this package?

navnavnav, 2022-10-09:
I've just pushed the changes to correct the missing header includes to the `develop` branch. If you build from that branch, those errors should not longer occur.

navnavnav, 2022-10-09:
BTW I just compiled Bloom with G++ 12.2.1 (on the `develop` branch) for the first time - it worked fine with those header include changes. I was also able to replicate the issue you had when building from `master` - so looks like G++ has become more strict with header includes, since v10.

shuyuan-liu, 2022-10-09:
Thanks @navnavnav for the quick fix! The `develop` branch compiles fine now for me with GCC 12.

> Oddly, I don't get this error when compiling with G++10.

I tried GCC 10, same result as yours. As you said it might have become stricter; not sure when or why though.

> Are you using CMake to trigger the build?

Yes I used the commands given by the main README, although I changed `Debug` to `Release`.

> may I ask why you're building from source?

Oh I didn't realise there were binary packages! I first searched the AUR and found [bloom-git](https://aur.archlinux.org/packages/bloom-git), which builds from source, and thought that was the only thing available. I tried your package just now and it worked, thanks!

closed 2022-10-09 by shuyuan-liu