bloombloombloom/Bloom #87: Fails to build with libprocps 4.0

jpf91, 2023-08-19:
Bloom unfortunately fails to build on Fedora 39, as that Fedora version ships the new libprocps 4.0. It seems like some header which is used in Bloom was removed. I'm not sure if there was also any API breakage which affects bloom or whether it's a simple matter of including other header files.

navnavnav, 2023-08-19:
Hey thanks for reporting this.

Do you know if Fedora 39 ships libprocps 3.*? Or does it only ship 4.0? Would be quite surprising if they've completely dropped version 3, as [version 4 was only released in March last year](https://gitlab.com/procps-ng/procps/-/releases/v4.0.0), and it does appear to contain breaking changes.

It will require some work to get Bloom working with that version. But I doubt it will be a lot of work. We only make use of that library in the [`ProcessService`](https://github.com/bloombloombloom/Bloom/blob/develop/src/Services/ProcessService.hpp), and its usage is fairly minimal.

I'll have a look at supporting the newer version, but not sure if I'll consider this a priority, as it really only affects those who build from source and can't obtain version 3. I'll post any updates on this ticket.

Thanks again for bringing this to my attention 👍🏽

closed 2023-12-31 by navnavnav

uliano, 2024-06-04:
I'm on debian, at first I tried to install binary but I get 

```
MESA: error: ZINK: failed to choose pdev
glx: failed to create drisw screen
```

so I tried to compile from source and I stumbled upon this libprocps thing

BTW according to the [wiki](https://gitlab.com/procps-ng/procps/-/wikis/newapi)  the new api is around since 9 years and it seems that old one will soon disappear from most distributions.

navnavnav, 2024-06-04:
```
MESA: error: ZINK: failed to choose pdev
glx: failed to create drisw screen
```
I don't think that's an issue with the binary. What version of Debian are you using? [Debian users seem to be having similar issues with other applications, with recent versions of Debian](https://www.google.com/search?q=%22MESA%3A+error%3A+ZINK%3A+failed+to+choose+pdev+glx%3A+failed+to+create+drisw+screen%22&oq=%22MESA%3A+error%3A+ZINK%3A+failed+to+choose+pdev+glx%3A+failed+to+create+drisw+screen%22). May be a driver issue. Building from source may not resolve this.

Is that the complete output? If not, can you include the whole output from Bloom? Does it continue running after that error, or does it abort?

RE libpocps, this is something I plan to support, but it's not a priority ATM. And no, the API has not been around for 9 years. They may have started working on it in 2015, but it was [released in March 2022](https://gitlab.com/procps-ng/procps/-/releases/v4.0.0). And was only made available via Debian's package repository from version 12 (Bookworm), in June last year.

uliano, 2024-06-04:
I'm not saying that building will solve all my issue, just trying to tackle somewhere.

I've tried on 2 different system, both running debian 13 (but the removal of the old library happened since 12, so my opinion is that may be worth considering to update the codebase) one within a wsl2 giving

```
uliano@Trolley2:~
❯ bloom init
MESA: error: ZINK: failed to choose pdev
glx: failed to create drisw screen
2024-06-04 14:49:39.689 CEST [MT]: [INFO] Bloom configuration file (bloom.yaml) created in working directory.
2024-06-04 14:49:39.689 CEST [MT]: [INFO] Shutting down Bloom
uliano@Trolley2:~
```
and the other one is on a headless desktop to which I logged using xrdp (which may be another source of confusion, I may find a monitor in a few days) giving
```
uliano@xlence2 ~
% bloom init
DRI3 not available
failed to load driver: zink
2024-06-04 14:51:05.312 CEST [MT]: [INFO] Bloom configuration file (bloom.yaml) created in working directory.
2024-06-04 14:51:05.312 CEST [MT]: [INFO] Shutting down Bloom
uliano@xlence2 ~
%
```
The fact that bloom quit may well be that these where just dumb test where I didn't set up anything just to see if I could run it.

navnavnav, 2024-06-04:
@uliano 

I've just tried this on Ubuntu 24.04 and I get the same error as you, but this is more of a warning it seems. Bloom runs fine even with that error message.

This also appears to be the case in your examples, where you ran `bloom init`. It's supposed to shutdown after creating the config file.

So the error should not prevent you from running Bloom.

I will look to update Bloom's use of libprocps at a later date. I have reopened this ticket and will post any updates here.

navnavnav, 2024-06-04:
Also @uliano, Debian 13 is not stable. It's still WIP. Unless your intention is to test Bloom on an unstable distro, I suggest you use a stable version.

SamsiFPV, 2025-03-23:
I tried to compile the newest version on Fedora 41, but it ships with procps-ng only, so it gives me the error:
```
fatal error: proc/readproc.h: No such file or directory
```

SamsiFPV, 2025-03-30:
@navnavnav any chance we can get a 2.0 rpm package somehow?