bloombloombloom/Bloom #97: Add support for UPDI over /dev/ttyUSBx and similar interfaces

kajusK, 2023-11-26:
The UPDI interface is simple enough to work with a cheap USB<>UART converter and a single resistor. Would it be possible to add option to use the tty device directly for debugging?

navnavnav, 2023-11-27:
The problem with reverse engineering is that you almost always end up with more questions than when you started. And when all you have is findings from reverse engineering, it's incredibly difficult to have any confidence in what you're building.

It will take a lot of time and effort to implement something like this to a point where it's stable and reliable. I just don't think it's worth it. But let's see - I will put this ticket into 'in consideration'. If it gets at least 10 👍🏽 reactions, I will reconsider.

But that other project looks interesting and seems to be fairly active, so maybe let them have a go at it.

kajusK, 2024-01-15:
Just a side note, the updi-gdbserver works quite nice for debugging my target, it would be even better if I could use the Bloom Insight by connecting to gdb server started by updi-gdbserver instead of launching the bloom gdb server which requires a proprietary debugger device. The Insight looks like a very nice tool for inspecting the MCU state, reading registers, etc. It's a pity it only works with the bloom gdb server while there are other gdb implementation out there.

Btw, the authors of updi-gdbserver are very nice people, it was a very pleasant experience debugging an issue on my board together with them.

felias-fogg, 2025-03-27:
Hi @kajusK, 

A year has gone by since you posted this message. 

Let me just point out that Microchip sells its SNAP debugger now for US-$ 14, which is still a bit more expensive than a USB-UART converter, but I am not sure whether it is worth the effort to spend time to reverse engineer things given this price -- except if you like reverse enginieering a lot ;-).

kajusK, 2025-06-06:
Thanks for pointing out, the SNAP looks great!

The udpi-gdbserver seems to be a dead project now, the last update is almost a year old, so I guess this issue can be closed....

closed 2025-06-06 by kajusK