bloombloombloom/Bloom #106: Raspberry Pi OS Version

lkoepsel, 2024-04-14:
Would it be possible to add a 64-bit ARM version? I'm attempting to setup a complete debugging suite on a Raspberry Pi 4 and just realized I'm unable to use Bloom.

Thanks,
Lief

navnavnav, 2024-04-14:
Hey Lief,

This is something I've been meaning to do.

It will be some time before I push the next release, so I won't be able to publish a new official build just yet. However, I will try to cross-compile an aarch64 build for you at some point in the next few days.

navnavnav, 2024-04-17:
Hey Lief,

Quick update on this:

I'm making progress but slowly. Cross-compiling a Qt application is a huge undertaking, because it requires compiling all recursive dependencies. Bloom doesn't have many dependencies, but Qt is one of them, and Qt has many of its own dependencies (OpenGL, X11, XCB and many other XOrg libraries).

I'm almost done with compiling Qt, so that's the most significant piece almost out of the way. Once that's done, just a few small libraries to compile and then I should be good to compile Bloom.

Hoping to have something for you by the end of this week.

At some point, I will move the build process to GitHub Actions, which will allow for the building for multiple architectures without cross-compiling. That will certainly make things easier, but I don't see it happening any time soon.

navnavnav, 2024-04-19:
Hey Lief,

The new aarch64 build is ready:

DEB: https://github.com/bloombloombloom/Bloom/releases/download/v1.0.0/Bloom-1.0.0-Linux-aarch64-unofficial.deb

The Raspberry Pi OS is Debian based, so a DEB package should suffice.

Few points:

- I have attached this build to the [v1.0.0 release](https://github.com/bloombloombloom/Bloom/releases/tag/v1.0.0) on GitHub, meaning the download link will not expire.
- This build will only work on the 64-bit version of the Raspberry Pi OS (it's a 64-bit binary).
- You'll need to be on a fairly recent version of the Raspberry Pi OS. I've tested the build with the most recent OS version available on the Raspberry Pi website. It may work on older versions.
- The Insight GUI wasn't designed to run on such a resource-constrained device. It worked fine during my testing but it was a little slow. I would advise only activating the Insight GUI when you really need it (keep `activateOnStartup` set to `false`).
- This is an unofficial build so I won't be making it available via the Bloom website. I do want to host an aarch64 build at some point, but there's a lot of work to do before I can do that, so it won't happen any time soon.

Please let me know if it works as expected on your Raspberry Pi.

lkoepsel, 2024-04-19:
Thank you!

I have demos the next few days, however I believe I can work on it on Sunday!

I’m looking forward to it!

I’ll keep track of all issues and successes and advise. 
Lief

lkoepsel, 2024-04-19:
I attempted to install and got the following. I wanted to keep the formatting, so it’s attached as a markdown file.

I wasn’t sure if you wanted me to post anything on GitHub.

Lief￼

navnavnav, 2024-04-19:
I don't see any attachment in your comment. You may have to submit the comment via the GitHub website. I'm happy for you to share any error logs/messages on GitHub 👍🏽

lkoepsel, 2024-04-19:
I downloaded and attempted to install on the latest version of Raspberry Pi OS...

```bash
sudo apt install ./Bloom-1.0.0-Linux-aarch64-unofficial.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'bloom' instead of './Bloom-1.0.0-Linux-aarch64-unofficial.deb'
The following packages were automatically installed and are no longer required:
  libcamera0.1 linux-headers-6.1.0-rpi7-common-rpi linux-headers-6.1.0-rpi7-rpi-2712
  linux-headers-6.1.0-rpi7-rpi-v8 linux-image-6.1.0-rpi7-rpi-2712 linux-image-6.1.0-rpi7-rpi-v8
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libb2-1 libcolord2 libegl-mesa0 libegl1 libepoxy0 libevdev2 libgbm1 libgtk-3-0 libgtk-3-bin
  libgtk-3-common libinput-bin libinput10 libmd4c0 libmtdev1 libopengl0 libproxy1v5 libqt6core6
  libqt6dbus6 libqt6gui6 libqt6network6 libqt6opengl6 libqt6openglwidgets6 libqt6widgets6 libqt6xml6
  libts0 libwacom-common libwacom9 libwayland-cursor0 libwayland-egl1 libwayland-server0
  libxcb-icccm4 libxcb-image0 libxcb-keysyms1 libxcb-render-util0 libxcb-util1 libxcb-xkb1
  libxkbcommon-x11-0 libxkbcommon0 qt6-gtk-platformtheme qt6-qpa-plugins qt6-translations-l10n
Suggested packages:
  colord gvfs libwacom-bin
The following NEW packages will be installed:
  bloom libb2-1 libcolord2 libegl-mesa0 libegl1 libepoxy0 libevdev2 libgbm1 libgtk-3-0 libgtk-3-bin
  libgtk-3-common libinput-bin libinput10 libmd4c0 libmtdev1 libopengl0 libproxy1v5 libqt6core6
  libqt6dbus6 libqt6gui6 libqt6network6 libqt6opengl6 libqt6openglwidgets6 libqt6widgets6 libqt6xml6
  libts0 libwacom-common libwacom9 libwayland-cursor0 libwayland-egl1 libwayland-server0
  libxcb-icccm4 libxcb-image0 libxcb-keysyms1 libxcb-render-util0 libxcb-util1 libxcb-xkb1
  libxkbcommon-x11-0 libxkbcommon0 qt6-gtk-platformtheme qt6-qpa-plugins qt6-translations-l10n
0 upgraded, 42 newly installed, 0 to remove and 0 not upgraded.
After this operation, 90.9 MB of additional disk space will be used.
Need to get 17.5 MB/20.3 MB of archives.
Do you want to continue? [Y/n]
Get:1 /home/lkoepsel/Bloom-1.0.0-Linux-aarch64-unofficial.deb bloom arm64 1.0.0 [2,803 kB]
Get:2 http://deb.debian.org/debian bookworm/main arm64 libb2-1 arm64 0.98.1-1.1 [16.1 kB]
Get:3 http://deb.debian.org/debian bookworm/main arm64 libqt6core6 arm64 6.4.2+dfsg-10 [1,468 kB]
Get:4 http://deb.debian.org/debian bookworm/main arm64 libegl1 arm64 1.6.0-1 [33.1 kB]
Get:5 http://deb.debian.org/debian bookworm/main arm64 libevdev2 arm64 1.13.0+dfsg-1 [32.0 kB]
Get:6 http://deb.debian.org/debian bookworm/main arm64 libmtdev1 arm64 1.1.6-1 [22.8 kB]
Get:7 http://deb.debian.org/debian bookworm/main arm64 libwacom-common all 2.6.0-1 [58.4 kB]
Get:8 http://deb.debian.org/debian bookworm/main arm64 libwacom9 arm64 2.6.0-1 [20.7 kB]
Get:9 http://deb.debian.org/debian bookworm/main arm64 libinput-bin arm64 1.22.1-1 [23.0 kB]
Get:10 http://deb.debian.org/debian bookworm/main arm64 libinput10 arm64 1.22.1-1 [112 kB]
Get:11 http://deb.debian.org/debian bookworm/main arm64 libmd4c0 arm64 0.4.8-1 [41.6 kB]
Get:12 http://deb.debian.org/debian bookworm/main arm64 libopengl0 arm64 1.6.0-1 [31.7 kB]
Get:13 http://deb.debian.org/debian bookworm/main arm64 libqt6dbus6 arm64 6.4.2+dfsg-10 [230 kB]
Get:14 http://deb.debian.org/debian bookworm/main arm64 libts0 arm64 1.22-1+b1 [63.4 kB]
Get:15 http://deb.debian.org/debian bookworm/main arm64 libxcb-icccm4 arm64 0.4.1-1.1 [26.9 kB]
Get:16 http://deb.debian.org/debian bookworm/main arm64 libxcb-util1 arm64 0.4.0-1+b1 [23.2 kB]
Get:17 http://deb.debian.org/debian bookworm/main arm64 libxcb-image0 arm64 0.4.0-2 [22.4 kB]
Get:18 http://deb.debian.org/debian bookworm/main arm64 libxcb-keysyms1 arm64 0.4.0-1+b2 [16.0 kB]
Get:19 http://deb.debian.org/debian bookworm/main arm64 libxcb-render-util0 arm64 0.3.9-1+b1 [17.5 kB]
Get:20 http://deb.debian.org/debian bookworm/main arm64 libxcb-xkb1 arm64 1.15-1 [128 kB]
Get:21 http://deb.debian.org/debian bookworm/main arm64 libxkbcommon0 arm64 1.5.0-1 [100 kB]
Get:22 http://deb.debian.org/debian bookworm/main arm64 libxkbcommon-x11-0 arm64 1.5.0-1 [14.6 kB]
Get:23 http://deb.debian.org/debian bookworm/main arm64 libqt6gui6 arm64 6.4.2+dfsg-10 [2,598 kB]
Get:24 http://deb.debian.org/debian bookworm/main arm64 libproxy1v5 arm64 0.4.18-1.2 [50.9 kB]
Get:25 http://deb.debian.org/debian bookworm/main arm64 libqt6network6 arm64 6.4.2+dfsg-10 [643 kB]
Get:26 http://deb.debian.org/debian bookworm/main arm64 libqt6widgets6 arm64 6.4.2+dfsg-10 [2,312 kB]
Get:27 http://deb.debian.org/debian bookworm/main arm64 libqt6opengl6 arm64 6.4.2+dfsg-10 [367 kB]
Get:28 http://deb.debian.org/debian bookworm/main arm64 libqt6openglwidgets6 arm64 6.4.2+dfsg-10 [43.0 kB]
Get:29 http://deb.debian.org/debian bookworm/main arm64 libqt6xml6 arm64 6.4.2+dfsg-10 [74.7 kB]
Get:30 http://deb.debian.org/debian bookworm/main arm64 libcolord2 arm64 1.4.6-2.2 [126 kB]
Get:31 http://deb.debian.org/debian bookworm/main arm64 libepoxy0 arm64 1.5.10-1 [198 kB]
Get:32 http://deb.debian.org/debian bookworm/main arm64 qt6-gtk-platformtheme arm64 6.4.2+dfsg-10 [55.5 kB]
Get:33 http://deb.debian.org/debian bookworm/main arm64 qt6-qpa-plugins arm64 6.4.2+dfsg-10 [84.2 kB]
Get:34 http://deb.debian.org/debian bookworm/main arm64 qt6-translations-l10n all 6.4.2-1 [1,647 kB]
Get:35 http://archive.raspberrypi.com/debian bookworm/main arm64 libwayland-server0 arm64 1.22.0-2.1~bpo12+rpt1 [31.9 kB]
[... cut ...]
Setting up libgtk-3-0:arm64 (1:3.24.38-2~deb12u1+rpt4+rpi1) ...
Setting up libgtk-3-bin (1:3.24.38-2~deb12u1+rpt4+rpi1) ...###################################....]
Setting up qt6-gtk-platformtheme:arm64 (6.4.2+dfsg-10) ...#####################################...]
Processing triggers for libc-bin (2.36-9+rpt2+deb12u4) ...######################################..]
N: Download is performed unsandboxed as root as file '/home/lkoepsel/Bloom-1.0.0-Linux-aarch64-unofficial.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
```

navnavnav, 2024-04-19:
That looks like it worked. I believe that "sandbox" error is just a
consequence of installing a local DEB file. Can you try running Bloom?

lkoepsel, 2024-04-19:
Well...that was almost embarrassing. :)

Unfortunately, bloom doesn't launch. I tried a couple of different variations:

```bash
bloom
bloom: error while loading shared libraries: libQt6UiTools.so.6: cannot open shared object file: No such file or directory

bloom init
bloom: error while loading shared libraries: libQt6UiTools.so.6: cannot open shared object file: No such file or directory

bloom --version
bloom: error while loading shared libraries: libQt6UiTools.so.6: cannot open shared object file: No such file or directory
```

navnavnav, 2024-04-19:
Ah ok, looks like I've missed some dependencies. Can you run `sudo apt install libqt6uitools6`.

It's possible that I've missed some other dependencies. If you get a similar error after running the above command, please run `ldd /opt/bloom/bin/bloom` and paste the output here.

lkoepsel, 2024-04-19:
I installed `libqt6...` and got the following error:
```bash
N: Download is performed unsandboxed as root as file '/home/lkoepsel/Bloom-1.0.0-Linux-aarch64-unofficial.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
```
So I ran ldd...per your request:
```bash
ldd /opt/bloom/bin/bloom
	linux-vdso.so.1 (0x0000007fb5ccd000)
	libQt6UiTools.so.6 => /lib/aarch64-linux-gnu/libQt6UiTools.so.6 (0x0000007fb58c0000)
	libusb-1.0.so.0 => /opt/bloom/bin/../lib/libusb-1.0.so.0 (0x0000007fb5890000)
	libhidapi-libusb.so.0 => /opt/bloom/bin/../lib/libhidapi-libusb.so.0 (0x0000007fb5860000)
	libprocps.so.8 => /opt/bloom/bin/../lib/libprocps.so.8 (0x0000007fb5810000)
	libyaml-cpp.so.0.7 => /opt/bloom/bin/../lib/libyaml-cpp.so.0.7 (0x0000007fb57a0000)
	libQt6Xml.so.6 => /lib/aarch64-linux-gnu/libQt6Xml.so.6 (0x0000007fb5750000)
	libQt6Network.so.6 => /lib/aarch64-linux-gnu/libQt6Network.so.6 (0x0000007fb55a0000)
	libQt6Svg.so.6 => not found
	libQt6Widgets.so.6 => /lib/aarch64-linux-gnu/libQt6Widgets.so.6 (0x0000007fb4e70000)
	libQt6Gui.so.6 => /lib/aarch64-linux-gnu/libQt6Gui.so.6 (0x0000007fb4770000)
	libQt6Core.so.6 => /lib/aarch64-linux-gnu/libQt6Core.so.6 (0x0000007fb4280000)
	libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007fb4060000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007fb3fc0000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007fb3f80000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007fb3dd0000)
	libQt6OpenGLWidgets.so.6 => /lib/aarch64-linux-gnu/libQt6OpenGLWidgets.so.6 (0x0000007fb3da0000)
	/lib/ld-linux-aarch64.so.1 (0x0000007fb5c90000)
	libudev.so.1 => /lib/aarch64-linux-gnu/libudev.so.1 (0x0000007fb3d50000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007fb3d20000)
	libsystemd.so.0 => /lib/aarch64-linux-gnu/libsystemd.so.0 (0x0000007fb3c30000)
	libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007fb3c00000)
	libgssapi_krb5.so.2 => /lib/aarch64-linux-gnu/libgssapi_krb5.so.2 (0x0000007fb3b90000)
	libbrotlidec.so.1 => /lib/aarch64-linux-gnu/libbrotlidec.so.1 (0x0000007fb3b60000)
	libzstd.so.1 => /lib/aarch64-linux-gnu/libzstd.so.1 (0x0000007fb3aa0000)
	libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007fb3a60000)
	libproxy.so.1 => /lib/aarch64-linux-gnu/libproxy.so.1 (0x0000007fb3a20000)
	libEGL.so.1 => /lib/aarch64-linux-gnu/libEGL.so.1 (0x0000007fb39e0000)
	libfontconfig.so.1 => /lib/aarch64-linux-gnu/libfontconfig.so.1 (0x0000007fb3970000)
	libX11.so.6 => /lib/aarch64-linux-gnu/libX11.so.6 (0x0000007fb3810000)
	libglib-2.0.so.0 => /lib/aarch64-linux-gnu/libglib-2.0.so.0 (0x0000007fb36b0000)
	libQt6DBus.so.6 => /lib/aarch64-linux-gnu/libQt6DBus.so.6 (0x0000007fb35e0000)
	libxkbcommon.so.0 => /lib/aarch64-linux-gnu/libxkbcommon.so.0 (0x0000007fb3570000)
	libGLX.so.0 => /lib/aarch64-linux-gnu/libGLX.so.0 (0x0000007fb3510000)
	libOpenGL.so.0 => /lib/aarch64-linux-gnu/libOpenGL.so.0 (0x0000007fb3490000)
	libpng16.so.16 => /lib/aarch64-linux-gnu/libpng16.so.16 (0x0000007fb3430000)
	libharfbuzz.so.0 => /lib/aarch64-linux-gnu/libharfbuzz.so.0 (0x0000007fb3310000)
	libmd4c.so.0 => /lib/aarch64-linux-gnu/libmd4c.so.0 (0x0000007fb32f0000)
	libfreetype.so.6 => /lib/aarch64-linux-gnu/libfreetype.so.6 (0x0000007fb3210000)
	libicui18n.so.72 => /lib/aarch64-linux-gnu/libicui18n.so.72 (0x0000007fb2ed0000)
	libicuuc.so.72 => /lib/aarch64-linux-gnu/libicuuc.so.72 (0x0000007fb2cb0000)
	libdouble-conversion.so.3 => /lib/aarch64-linux-gnu/libdouble-conversion.so.3 (0x0000007fb2c70000)
	libb2.so.1 => /lib/aarch64-linux-gnu/libb2.so.1 (0x0000007fb2c50000)
	libpcre2-16.so.0 => /lib/aarch64-linux-gnu/libpcre2-16.so.0 (0x0000007fb2bb0000)
	libQt6OpenGL.so.6 => /lib/aarch64-linux-gnu/libQt6OpenGL.so.6 (0x0000007fb2af0000)
	libcap.so.2 => /lib/aarch64-linux-gnu/libcap.so.2 (0x0000007fb2ac0000)
	libgcrypt.so.20 => /lib/aarch64-linux-gnu/libgcrypt.so.20 (0x0000007fb29b0000)
	liblzma.so.5 => /lib/aarch64-linux-gnu/liblzma.so.5 (0x0000007fb2960000)
	liblz4.so.1 => /lib/aarch64-linux-gnu/liblz4.so.1 (0x0000007fb2910000)
	libkrb5.so.3 => /lib/aarch64-linux-gnu/libkrb5.so.3 (0x0000007fb2820000)
	libk5crypto.so.3 => /lib/aarch64-linux-gnu/libk5crypto.so.3 (0x0000007fb27d0000)
	libcom_err.so.2 => /lib/aarch64-linux-gnu/libcom_err.so.2 (0x0000007fb27a0000)
	libkrb5support.so.0 => /lib/aarch64-linux-gnu/libkrb5support.so.0 (0x0000007fb2770000)
	libbrotlicommon.so.1 => /lib/aarch64-linux-gnu/libbrotlicommon.so.1 (0x0000007fb2730000)
	libGLdispatch.so.0 => /lib/aarch64-linux-gnu/libGLdispatch.so.0 (0x0000007fb25a0000)
	libexpat.so.1 => /lib/aarch64-linux-gnu/libexpat.so.1 (0x0000007fb2550000)
	libxcb.so.1 => /lib/aarch64-linux-gnu/libxcb.so.1 (0x0000007fb2500000)
	libpcre2-8.so.0 => /lib/aarch64-linux-gnu/libpcre2-8.so.0 (0x0000007fb2450000)
	libdbus-1.so.3 => /lib/aarch64-linux-gnu/libdbus-1.so.3 (0x0000007fb23d0000)
	libgraphite2.so.3 => /lib/aarch64-linux-gnu/libgraphite2.so.3 (0x0000007fb2390000)
	libicudata.so.72 => /lib/aarch64-linux-gnu/libicudata.so.72 (0x0000007fb05a0000)
	libgomp.so.1 => /lib/aarch64-linux-gnu/libgomp.so.1 (0x0000007fb0530000)
	libgpg-error.so.0 => /lib/aarch64-linux-gnu/libgpg-error.so.0 (0x0000007fb04e0000)
	libkeyutils.so.1 => /lib/aarch64-linux-gnu/libkeyutils.so.1 (0x0000007fb04b0000)
	libresolv.so.2 => /lib/aarch64-linux-gnu/libresolv.so.2 (0x0000007fb0480000)
	libXau.so.6 => /lib/aarch64-linux-gnu/libXau.so.6 (0x0000007fb0460000)
	libXdmcp.so.6 => /lib/aarch64-linux-gnu/libXdmcp.so.6 (0x0000007fb0440000)
	libbsd.so.0 => /lib/aarch64-linux-gnu/libbsd.so.0 (0x0000007fb0400000)
	libmd.so.0 => /lib/aarch64-linux-gnu/libmd.so.0 (0x0000007fb03d0000)
```

navnavnav, 2024-04-19:
ok, just one missing dependency left: `sudo apt-get install libqt6svg`. That should do it. Then try running Bloom - it should work.

I neglected to properly inspect the binary's dependencies when I prepared the DEB package (it was very late at night - was exhausted), so I failed to list those packages as dependencies in the DEB control file.

navnavnav, 2024-04-19:
> and got the following error

Don't worry about that error. I see it a fair bit when installing debian packages. I spent some time looking into it a while ago - IIRC the internet said it wasn't really an issue - more like a warning.

lkoepsel, 2024-04-19:
```bash
sudo apt install libqt6svg
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libqt6svg
```

navnavnav, 2024-04-19:
Ugh...sorry Lief, typo from me. It should be "libqt6svg6". So `sudo apt install libqt6svg6`

lkoepsel, 2024-04-19:
No problem, completely understand. On to the next one!

```
bloom
qt.qpa.xcb: could not connect to display
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: linuxfb, offscreen, minimalegl, minimal, vkkhrdisplay, eglfs, vnc, xcb.

Aborted
```

lkoepsel, 2024-04-19:
Just realized, I'm attempting to run this headless...which probably won't? work?

Could that be the issue with the "could not connect to the display"? I'm ssh'ing into the Raspberry Pi...

navnavnav, 2024-04-19:
> Could that be the issue with the "could not connect to the display"? I'm ssh'ing into the Raspberry Pi...

Hmm, it could be. Did you install the "Lite" version of Pi OS? Do you have a desktop environment installed on your Pi?

Can you try running `ldd /usr/lib/aarch64-linux-gnu/qt6/plugins/platforms/libqxcb.so` and pasting the output here?

lkoepsel, 2024-04-19:
1) I did install the Lite version, apologies if this is the issue. I can install and test on full version if required, though I wanted to go headless...maybe I vnc? rdp?
2) ```
ldd /usr/lib/aarch64-linux-gnu/qt6/plugins/platforms/libqxcb.so
	linux-vdso.so.1 (0x0000007f809ff000)
	libQt6XcbQpa.so.6 => /lib/aarch64-linux-gnu/libQt6XcbQpa.so.6 (0x0000007f808e0000)
	libQt6Gui.so.6 => /lib/aarch64-linux-gnu/libQt6Gui.so.6 (0x0000007f801e0000)
	libQt6Core.so.6 => /lib/aarch64-linux-gnu/libQt6Core.so.6 (0x0000007f7fcf0000)
	libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007f7fad0000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f7f920000)
	libxcb-icccm.so.4 => /lib/aarch64-linux-gnu/libxcb-icccm.so.4 (0x0000007f7f900000)
	libxcb-image.so.0 => /lib/aarch64-linux-gnu/libxcb-image.so.0 (0x0000007f7f8e0000)
	libxcb-keysyms.so.1 => /lib/aarch64-linux-gnu/libxcb-keysyms.so.1 (0x0000007f7f8c0000)
	libxcb-randr.so.0 => /lib/aarch64-linux-gnu/libxcb-randr.so.0 (0x0000007f7f890000)
	libxcb-render.so.0 => /lib/aarch64-linux-gnu/libxcb-render.so.0 (0x0000007f7f860000)
	libxcb-render-util.so.0 => /lib/aarch64-linux-gnu/libxcb-render-util.so.0 (0x0000007f7f840000)
	libxcb-shape.so.0 => /lib/aarch64-linux-gnu/libxcb-shape.so.0 (0x0000007f7f810000)
	libxcb-shm.so.0 => /lib/aarch64-linux-gnu/libxcb-shm.so.0 (0x0000007f7f7e0000)
	libxcb-sync.so.1 => /lib/aarch64-linux-gnu/libxcb-sync.so.1 (0x0000007f7f7b0000)
	libxcb.so.1 => /lib/aarch64-linux-gnu/libxcb.so.1 (0x0000007f7f760000)
	libxcb-xfixes.so.0 => /lib/aarch64-linux-gnu/libxcb-xfixes.so.0 (0x0000007f7f730000)
	libxcb-xkb.so.1 => /lib/aarch64-linux-gnu/libxcb-xkb.so.1 (0x0000007f7f6f0000)
	libxkbcommon.so.0 => /lib/aarch64-linux-gnu/libxkbcommon.so.0 (0x0000007f7f680000)
	libglib-2.0.so.0 => /lib/aarch64-linux-gnu/libglib-2.0.so.0 (0x0000007f7f520000)
	libX11-xcb.so.1 => /lib/aarch64-linux-gnu/libX11-xcb.so.1 (0x0000007f7f4f0000)
	libX11.so.6 => /lib/aarch64-linux-gnu/libX11.so.6 (0x0000007f7f390000)
	libSM.so.6 => /lib/aarch64-linux-gnu/libSM.so.6 (0x0000007f7f370000)
	libICE.so.6 => /lib/aarch64-linux-gnu/libICE.so.6 (0x0000007f7f340000)
	libxkbcommon-x11.so.0 => /lib/aarch64-linux-gnu/libxkbcommon-x11.so.0 (0x0000007f7f310000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f7f270000)
	/lib/ld-linux-aarch64.so.1 (0x0000007f809c2000)
	libEGL.so.1 => /lib/aarch64-linux-gnu/libEGL.so.1 (0x0000007f7f230000)
	libfontconfig.so.1 => /lib/aarch64-linux-gnu/libfontconfig.so.1 (0x0000007f7f1c0000)
	libQt6DBus.so.6 => /lib/aarch64-linux-gnu/libQt6DBus.so.6 (0x0000007f7f0f0000)
	libGLX.so.0 => /lib/aarch64-linux-gnu/libGLX.so.0 (0x0000007f7f090000)
	libOpenGL.so.0 => /lib/aarch64-linux-gnu/libOpenGL.so.0 (0x0000007f7f010000)
	libpng16.so.16 => /lib/aarch64-linux-gnu/libpng16.so.16 (0x0000007f7efb0000)
	libharfbuzz.so.0 => /lib/aarch64-linux-gnu/libharfbuzz.so.0 (0x0000007f7ee90000)
	libmd4c.so.0 => /lib/aarch64-linux-gnu/libmd4c.so.0 (0x0000007f7ee70000)
	libfreetype.so.6 => /lib/aarch64-linux-gnu/libfreetype.so.6 (0x0000007f7ed90000)
	libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007f7ed50000)
	libicui18n.so.72 => /lib/aarch64-linux-gnu/libicui18n.so.72 (0x0000007f7ea10000)
	libicuuc.so.72 => /lib/aarch64-linux-gnu/libicuuc.so.72 (0x0000007f7e7f0000)
	libdouble-conversion.so.3 => /lib/aarch64-linux-gnu/libdouble-conversion.so.3 (0x0000007f7e7b0000)
	libb2.so.1 => /lib/aarch64-linux-gnu/libb2.so.1 (0x0000007f7e790000)
	libpcre2-16.so.0 => /lib/aarch64-linux-gnu/libpcre2-16.so.0 (0x0000007f7e6f0000)
	libzstd.so.1 => /lib/aarch64-linux-gnu/libzstd.so.1 (0x0000007f7e630000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007f7e5f0000)
	libxcb-util.so.1 => /lib/aarch64-linux-gnu/libxcb-util.so.1 (0x0000007f7e5d0000)
	libXau.so.6 => /lib/aarch64-linux-gnu/libXau.so.6 (0x0000007f7e5b0000)
	libXdmcp.so.6 => /lib/aarch64-linux-gnu/libXdmcp.so.6 (0x0000007f7e590000)
	libpcre2-8.so.0 => /lib/aarch64-linux-gnu/libpcre2-8.so.0 (0x0000007f7e4e0000)
	libuuid.so.1 => /lib/aarch64-linux-gnu/libuuid.so.1 (0x0000007f7e4b0000)
	libbsd.so.0 => /lib/aarch64-linux-gnu/libbsd.so.0 (0x0000007f7e470000)
	libGLdispatch.so.0 => /lib/aarch64-linux-gnu/libGLdispatch.so.0 (0x0000007f7e2e0000)
	libexpat.so.1 => /lib/aarch64-linux-gnu/libexpat.so.1 (0x0000007f7e290000)
	libdbus-1.so.3 => /lib/aarch64-linux-gnu/libdbus-1.so.3 (0x0000007f7e210000)
	libgraphite2.so.3 => /lib/aarch64-linux-gnu/libgraphite2.so.3 (0x0000007f7e1d0000)
	libbrotlidec.so.1 => /lib/aarch64-linux-gnu/libbrotlidec.so.1 (0x0000007f7e1a0000)
	libicudata.so.72 => /lib/aarch64-linux-gnu/libicudata.so.72 (0x0000007f7c3b0000)
	libgomp.so.1 => /lib/aarch64-linux-gnu/libgomp.so.1 (0x0000007f7c340000)
	libmd.so.0 => /lib/aarch64-linux-gnu/libmd.so.0 (0x0000007f7c310000)
	libsystemd.so.0 => /lib/aarch64-linux-gnu/libsystemd.so.0 (0x0000007f7c220000)
	libbrotlicommon.so.1 => /lib/aarch64-linux-gnu/libbrotlicommon.so.1 (0x0000007f7c1e0000)
	libcap.so.2 => /lib/aarch64-linux-gnu/libcap.so.2 (0x0000007f7c1b0000)
	libgcrypt.so.20 => /lib/aarch64-linux-gnu/libgcrypt.so.20 (0x0000007f7c0a0000)
	liblzma.so.5 => /lib/aarch64-linux-gnu/liblzma.so.5 (0x0000007f7c050000)
	liblz4.so.1 => /lib/aarch64-linux-gnu/liblz4.so.1 (0x0000007f7c000000)
	libgpg-error.so.0 => /lib/aarch64-linux-gnu/libgpg-error.so.0 (0x0000007f7bfb0000)
```

navnavnav, 2024-04-19:
Ah I see. So just to clarify: you're aware that you can't run any GUI apps on the "lite" Pi OS, as there's no desktop environment, right? So you won't be able to use the Insight GUI at all. Is that OK/expected?

If so, then it sounds like what you want is a Bloom build with the Insight GUI stripped out. As it happens, I'm currently in the process of building that as it's something I've been meaning to test on my Pi. The build should be ready in about 20 mins. Will share it with you once it's done.

The ldd output looks fine. So yeah, my guess is that the "lite" OS version is missing some other dependencies that Qt requires.

Give me about 20 mins and I'll have the "Bloom Light" build ready 👍🏽

lkoepsel, 2024-04-19:
Thanks! And I'll try both, the Full and Lite versions, thought it will have to be on Sunday...I've got to stop for now and get back to my household chores. :) I'll be out all day tomorrow and will get back on this Sunday AM. I'm excited to test it out.

Yes, I'm aware that Insight would not be available. I was planning on having the students query memory etc via gdb.

I do want to explore using the GUI version with Insight as you've done such a nice job of the design and look, its a shame not to use it. I believe it might be easier for students (college) to understand.

navnavnav, 2024-04-20:
>  Yes, I'm aware that Insight would not be available. I was planning on having the students query memory etc via gdb.

Cool, that makes sense.

I have prepared two packages for you:

### Bloom v1.0.0 (full version - with Insight GUI):
DEB: https://github.com/bloombloombloom/Bloom/releases/download/v1.0.0/Bloom-1.0.0-Linux-aarch64-unofficial.deb

Key points:

- This package contains the full version of Bloom (the Insight GUI has not been stripped out)
- It's pretty much the same as the package I provided yesterday, but with the missing dependencies added to the DEB control file
- To run it on your Pi, you'll need to install the full Pi OS (with the desktop environment - not the "Lite" version)
- Because this build includes the Insight GUI, the package will install many Qt GUI dependencies

### Bloom "Light" v1.0.0:
DEB: https://github.com/bloombloombloom/Bloom/releases/download/v1.0.0/Bloom-1.0.0-Light-Linux-aarch64-unofficial.deb

Key points:

- This package contains the "Light" version of Bloom - the Insight GUI has been completely removed from the build. This means no Insight functionality will be available with this build
- This package has zero GUI dependencies
- You should be able to run this on your Pi without installing a desktop environment, so it should work fine with the "Lite" version of Pi OS
- I highly recommend using this build on resource constrained hardware like the Pi. It has a much smaller memory and CPU footprint and so is much better suited IMO. In fact, the very purpose of this build configuration was so that users could [run Bloom on their Raspberry Pis](https://github.com/bloombloombloom/Bloom/issues/57)

>  and will get back on this Sunday AM

No problem! I'm in no rush. Take your time 👍🏽 

Please let me know how it goes. Hopefully you won't have many issues with the "Light" version, as the package comes with only a handful of dependencies.

navnavnav, 2024-04-20:
One more thing: be sure to remove the previously installed version before installing either of the packages above: `sudo apt remove bloom` 👍🏽

lkoepsel, 2024-04-21:
Status so far...
Headless works great!!!
Its been awhile since I worked with gdb/bloom, so it took a bit of reading old notes to get going. 
A couple of questions:

1) I'm using avrdude to manage DWEN, instead of bloom. Does that make sense? 

2) My config file is the following, do I need to add anything else?
```
environments:
  default:
    shutdownPostDebugSession: true

    tool:
      name: "snap"

    target:
      name: "atmega328p"
      physicalInterface: "debug-wire"
      hardwareBreakpoints: true

    server:
      name: "avr-gdb-rsp"
      ipAddress: "127.0.0.1"
      port: 1442

insight:
  activateOnStartup: false
```
3) I do miss insight, its so lovely and easy to use. :) 

I'll work on the full version, probably tomorrow. I need to do a bit of setup to use it properly.
Thanks!
Lief

lkoepsel, 2024-04-21:
I had previously set DWEN using the avrdude command:
```
avrdude  -p atmega328p -c snap_isp -P usb -U hfuse:w:0x9A:m
```
And everything worked well. I set the DWEN fuse back to 0, using:
```
avrdude  -p atmega328p -c snap_isp -P usb -U hfuse:w:0xDA:m
``` 
and started bloom with debuglogging on:
```
2024-04-21 11:45:26.936 PDT [MT]: [WARNING] The Insight component has been excluded from this build. All Insight related configuration parameters will be ignored.
2024-04-21 11:45:26.938 PDT [MT]: [DEBUG] Debug log printing has been enabled
2024-04-21 11:45:26.938 PDT [MT]: [DEBUG] Bloom version: 1.0.0
2024-04-21 11:45:26.938 PDT [MT]: [INFO] Selected environment: "default"
2024-04-21 11:45:26.938 PDT [MT]: [DEBUG] Number of environments extracted from config: 1
2024-04-21 11:45:26.939 PDT [SH]: [DEBUG] Starting SignalHandler
2024-04-21 11:45:26.939 PDT [SH]: [DEBUG] SignalHandler ready
2024-04-21 11:45:26.939 PDT [TC]: [INFO] Starting TargetController
2024-04-21 11:45:26.959 PDT [TC]: [INFO] Connecting to debug tool
2024-04-21 11:45:26.976 PDT [TC]: [DEBUG] HID device path: 1-1.4:1.0
2024-04-21 11:45:27.015 PDT [TC]: [INFO] Debug tool connected
2024-04-21 11:45:27.015 PDT [TC]: [INFO] Debug tool name: MPLAB Snap
2024-04-21 11:45:27.053 PDT [TC]: [INFO] Debug tool serial:
2024-04-21 11:45:27.057 PDT [TC]: [DEBUG] Loading AVR8 target description file: /opt/bloom/bin/../resources/TargetDescriptionFiles/AVR/AVR8/MEGA/ATMEGA328P.xml
2024-04-21 11:45:27.109 PDT [TC]: [INFO] Activating target
2024-04-21 11:45:27.147 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x00, id: 0x00, value: 0x01)
2024-04-21 11:45:27.185 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x00, id: 0x01, value: 0x02)
2024-04-21 11:45:27.223 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x01, id: 0x00, value: 0x05)
2024-04-21 11:45:27.223 PDT [TC]: [DEBUG] Setting FLASH_PAGE_SIZE AVR8 device parameter
2024-04-21 11:45:27.260 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x00, value: 0x8000)
2024-04-21 11:45:27.260 PDT [TC]: [DEBUG] Setting FLASH_SIZE AVR8 device parameter
2024-04-21 11:45:27.297 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x02, value: 0x00800000)
2024-04-21 11:45:27.297 PDT [TC]: [DEBUG] Setting FLASH_BASE AVR8 device parameter
2024-04-21 11:45:27.336 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x06, value: 0x00000000)
2024-04-21 11:45:27.336 PDT [TC]: [DEBUG] Setting SRAM_START AVR8 device parameter
2024-04-21 11:45:27.373 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x0e, value: 0x0001)
2024-04-21 11:45:27.373 PDT [TC]: [DEBUG] Setting EEPROM_SIZE AVR8 device parameter
2024-04-21 11:45:27.410 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x10, value: 0x0004)
2024-04-21 11:45:27.410 PDT [TC]: [DEBUG] Setting EEPROM_PAGE_SIZE AVR8 device parameter
2024-04-21 11:45:27.449 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x12, value: 0x04)
2024-04-21 11:45:27.449 PDT [TC]: [DEBUG] Setting OCD_REVISION AVR8 device parameter
2024-04-21 11:45:27.486 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x13, value: 0x01)
2024-04-21 11:45:27.486 PDT [TC]: [DEBUG] Setting OCD_DATA_REGISTER AVR8 device parameter
2024-04-21 11:45:27.523 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x18, value: 0x31)
2024-04-21 11:45:27.524 PDT [TC]: [DEBUG] Setting SPMCR_REGISTER AVR8 device parameter
2024-04-21 11:45:27.562 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x1d, value: 0x57)
2024-04-21 11:45:27.562 PDT [TC]: [DEBUG] Setting BOOT_START_ADDR AVR8 device parameter
2024-04-21 11:45:27.600 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x0a, value: 0x003f0000)
2024-04-21 11:45:27.600 PDT [TC]: [DEBUG] Setting OSCCAL_ADDR AVR8 device parameter
2024-04-21 11:45:27.638 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x1e, value: 0x46)
2024-04-21 11:45:27.638 PDT [TC]: [DEBUG] Setting EEARL_ADDR AVR8 device parameter
2024-04-21 11:45:27.675 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x1a, value: 0x21)
2024-04-21 11:45:27.676 PDT [TC]: [DEBUG] Setting EEARH_ADDR AVR8 device parameter
2024-04-21 11:45:27.714 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x19, value: 0x22)
2024-04-21 11:45:27.714 PDT [TC]: [DEBUG] Setting EECR_ADDR AVR8 device parameter
2024-04-21 11:45:27.752 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x1b, value: 0x1f)
2024-04-21 11:45:27.753 PDT [TC]: [DEBUG] Setting EEDR_ADDR AVR8 device parameter
2024-04-21 11:45:27.791 PDT [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x1c, value: 0x20)
2024-04-21 11:45:29.444 PDT [TC]: [DEBUG] Failed to activate physical interface on AVR8 target - retrying with external reset applied.
2024-04-21 11:45:31.099 PDT [TC]: [WARNING] Failed to activate the debugWire physical interface - attempting to access target via the ISP interface, for DWEN fuse bit inspection.
2024-04-21 11:45:31.099 PDT [TC]: [DEBUG] Extracting ISP parameters from TDF
2024-04-21 11:45:31.099 PDT [TC]: [INFO] Initiating ISP interface
2024-04-21 11:45:31.153 PDT [TC]: [WARNING] Updating the DWEN fuse bit is a potentially dangerous operation. Bloom is provided "AS IS", without warranty of any kind. You are using Bloom at your own risk. In no event shall the copyright owner or contributors be liable for any d [... cut ...]
2024-04-21 11:45:35.294 PDT [MT]: [DEBUG] Joining TargetController thread
2024-04-21 11:45:35.294 PDT [MT]: [DEBUG] TargetController thread joined
2024-04-21 11:45:35.294 PDT [MT]: [DEBUG] Joining SignalHandler thread
2024-04-21 11:45:35.295 PDT [SH]: [DEBUG] SIGNAL 10 received
2024-04-21 11:45:35.295 PDT [SH]: [INFO] Shutting down SignalHandler
2024-04-21 11:45:35.295 PDT [MT]: [DEBUG] SignalHandler thread joined
2024-04-21 11:45:35.295 PDT [MT]: [DEBUG] Saving project settings to /home/lkoepsel/AVR_C/examples/blink/.bloom/settings.json
```

navnavnav, 2024-04-21:
Ah, I think I see what's happened here. You used AVRDUDE to update the fuses, which can only be done via the ISP interface. But if the DWEN fuse bit is enabled, the ISP interface will become unavailable. So to get around this, AVRDUDE invokes a command to temporarily disable the debugWire interface, so that it can access the ISP interface.

By "temporarily", I mean until a power cycle. So this temporary process doesn't actually involve updating the DWEN fuse - it's just a command we can send to the debug tool to disable the debugWire interface until the next power cycle.

I suspect you ran Bloom right after using AVRDUDE to update fuses, but you didn't cycle the target power? If so, that would be the problem.

Also, just FYI, setting the fuse bit to 0 (clearing the bit), means programming the fuse. A set bit means "unprogrammed".

So what you need to do is set the fuse bit to 1, cycle target power, then run Bloom. From that point, Bloom should manage the DWEN just fine. Let me know how that goes.

lkoepsel, 2024-04-21:
You are correct, I followed the same steps, however, this time, I cycled power on the Uno. Everything works.

I'll leave manageDwenFuseBit: true and will turn off debugLogging. 

thx

lkoepsel, 2024-04-22:
Two things:
1) Had this error on `bloom init`:
```bash
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
QStandardPaths: wrong permissions on runtime directory /run/user/1000, 0770 instead of 0700
2024-04-22 07:18:20.207 PDT [MT]: [INFO] Bloom configuration file (bloom.yaml) created in working directory.
2024-04-22 07:18:20.208 PDT [MT]: [INFO] Shutting down Bloom
```
2) Got this:
[image]

Its a beautiful thing!!

I'll advise if I have any other issues, however, for now I think this is closed.

navnavnav, 2024-04-22:
Great to see you got the full build working.

> qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""

This is nothing to worry about - just a warning from the Qt framework. Feel free to ignore.

> QStandardPaths: wrong permissions on runtime directory /run/user/1000, 0770 instead of 0700

Not seen this before, though I suspect it's not a problem. May look into it later if does turn out to be an issue.

> I'll advise if I have any other issues, however, for now I think this is closed.

Excellent. Thanks for spending time on testing this, Lief. I will close this ticket now. Feel free to create another, if you face any other issues.

closed 2024-04-22 by navnavnav