bloombloombloom/Bloom #107: Wrong target signature (ATtiny416)

fakr00n, 2024-04-15:
I'm using the [ATTINY416 XPLAINED NANO](https://www.microchip.com/en-us/development-tool/attiny416-xnano) board. The board stopped working with bloom for some odd reason (it was working fine yesteraday).

Here is the bloom debug output

    2024-04-15 17:56:05.907 CEST [MT]: [DEBUG] Debug log printing has been enabled
    2024-04-15 17:56:05.907 CEST [MT]: [DEBUG] Bloom version: 1.0.0
    2024-04-15 17:56:05.907 CEST [MT]: [INFO] Selected environment: "default"
    2024-04-15 17:56:05.907 CEST [MT]: [DEBUG] Number of environments extracted from config: 1
    2024-04-15 17:56:05.907 CEST [SH]: [DEBUG] Starting SignalHandler
    2024-04-15 17:56:05.907 CEST [TC]: [INFO] Starting TargetController
    2024-04-15 17:56:05.907 CEST [SH]: [DEBUG] SignalHandler ready
    2024-04-15 17:56:05.934 CEST [TC]: [INFO] Connecting to debug tool
    2024-04-15 17:56:05.964 CEST [TC]: [DEBUG] HID device path: 7-2.4.4.2.2:1.0
    2024-04-15 17:56:06.009 CEST [TC]: [INFO] Debug tool connected
    2024-04-15 17:56:06.009 CEST [TC]: [INFO] Debug tool name: Xplained Nano
    2024-04-15 17:56:06.049 CEST [TC]: [INFO] Debug tool serial: ATML2795042700004593
    2024-04-15 17:56:06.050 CEST [TC]: [DEBUG] Loading AVR8 target description file: /opt/bloom/bin/../resources/TargetDescriptionFiles/AVR/AVR8/TINY/ATTINY416.xml
    2024-04-15 17:56:06.071 CEST [TC]: [DEBUG] Failed to extract memory segment from target description element - Unknown type: "other"
    2024-04-15 17:56:06.071 CEST [TC]: [DEBUG] Failed to extract memory segment from target description element - Unknown type: "user_signatures"
    2024-04-15 17:56:06.077 CEST [TC]: [INFO] Activating target
    2024-04-15 17:56:06.121 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x01, id: 0x31, value: 0x0807)
    2024-04-15 17:56:06.161 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x03, id: 0x06, value: 0x00)
    2024-04-15 17:56:06.201 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x00, id: 0x00, value: 0x05)
    2024-04-15 17:56:06.241 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x00, id: 0x01, value: 0x02)
    2024-04-15 17:56:06.281 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x01, id: 0x00, value: 0x08)
    2024-04-15 17:56:06.281 CEST [TC]: [DEBUG] Setting UPDI_PROGMEM_BASE_ADDR AVR8 device parameter
    2024-04-15 17:56:06.321 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x00, value: 0x0080)
    2024-04-15 17:56:06.321 CEST [TC]: [DEBUG] Setting UPDI_PROGMEM_BASE_ADDR_MSB AVR8 device parameter
    2024-04-15 17:56:06.361 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x2c, value: 0x00)
    2024-04-15 17:56:06.361 CEST [TC]: [DEBUG] Setting UPDI_FLASH_PAGE_SIZE AVR8 device parameter
    2024-04-15 17:56:06.401 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x02, value: 0x40)
    2024-04-15 17:56:06.401 CEST [TC]: [DEBUG] Setting UPDI_FLASH_PAGE_SIZE_MSB AVR8 device parameter
    2024-04-15 17:56:06.441 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x2d, value: 0x00)
    2024-04-15 17:56:06.441 CEST [TC]: [DEBUG] Setting UPDI_EEPROM_PAGE_SIZE AVR8 device parameter
    2024-04-15 17:56:06.481 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x03, value: 0x20)
    2024-04-15 17:56:06.481 CEST [TC]: [DEBUG] Setting UPDI_NVMCTRL_ADDR AVR8 device parameter
    2024-04-15 17:56:06.521 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x04, value: 0x0010)
    2024-04-15 17:56:06.521 CEST [TC]: [DEBUG] Setting UPDI_OCD_ADDR AVR8 device parameter
    2024-04-15 17:56:06.561 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x06, value: 0x800f)
    2024-04-15 17:56:06.561 CEST [TC]: [DEBUG] Setting UPDI_FLASH_SIZE AVR8 device parameter
    2024-04-15 17:56:06.601 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x12, value: 0x00100000)
    2024-04-15 17:56:06.601 CEST [TC]: [DEBUG] Setting UPDI_EEPROM_SIZE AVR8 device parameter
    2024-04-15 17:56:06.641 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x16, value: 0x8000)
    2024-04-15 17:56:06.641 CEST [TC]: [DEBUG] Setting UPDI_EEPROM_BASE_ADDR AVR8 device parameter
    2024-04-15 17:56:06.681 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x20, value: 0x0014)
    2024-04-15 17:56:06.681 CEST [TC]: [DEBUG] Setting UPDI_SIG_BASE_ADDR AVR8 device parameter
    2024-04-15 17:56:06.721 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x24, value: 0x0011)
    2024-04-15 17:56:06.721 CEST [TC]: [DEBUG] Setting UPDI_FUSE_BASE_ADDR AVR8 device parameter
    2024-04-15 17:56:06.761 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x26, value: 0x8012)
    2024-04-15 17:56:06.761 CEST [TC]: [DEBUG] Setting UPDI_FUSE_SIZE AVR8 device parameter
    2024-04-15 17:56:06.801 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x1a, value: 0x0a00)
    2024-04-15 17:56:06.801 CEST [TC]: [DEBUG] Setting UPDI_LOCK_BASE_ADDR AVR8 device parameter
    2024-04-15 17:56:06.841 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x28, value: 0x8a12)
    2024-04-15 17:56:06.881 CEST [TC]: [DEBUG] Setting AVR8 EDBG parameter (context: 0x02, id: 0x2e, value: 0x00)
    2024-04-15 17:56:07.369 CEST [TC]: [ERROR] The TargetController encountered a fatal error. See below for errors:
    2024-04-15 17:56:07.369 CEST [TC]: [ERROR] Failed to validate connected target - target signature mismatch.
    The target signature ("0x552400") does not match the AVR8 target description signature ("0x1e9221"). This will likely be due to an incorrect target name in the configuration file (bloom.yaml).
    2024-04-15 17:56:07.369 CEST [TC]: [INFO] Shutting down TargetController
    2024-04-15 17:56:07.369 CEST [TC]: [INFO] Deactivating target
    2024-04-15 17:56:07.529 CEST [TC]: [INFO] Closing debug tool
    2024-04-15 17:56:07.570 CEST [TC]: [DEBUG] Event "TargetControllerThreadStateChanged" (1) registered for listener ApplicationEventListener
    2024-04-15 17:56:07.570 CEST [MT]: [ERROR] TargetController failed to start up
    2024-04-15 17:56:07.570 CEST [MT]: [INFO] Shutting down Bloom
    2024-04-15 17:56:07.570 CEST [MT]: [DEBUG] Joining TargetController thread
    2024-04-15 17:56:07.570 CEST [MT]: [DEBUG] TargetController thread joined
    2024-04-15 17:56:07.570 CEST [MT]: [DEBUG] Joining SignalHandler thread
    2024-04-15 17:56:07.570 CEST [SH]: [DEBUG] SIGNAL 10 received
    2024-04-15 17:56:07.570 CEST [SH]: [INFO] Shutting down SignalHandler
    2024-04-15 17:56:07.571 CEST [MT]: [DEBUG] SignalHandler thread joined
    2024-04-15 17:56:07.571 CEST [MT]: [DEBUG] Saving project settings to /home/fakr00n/Documents/git/develop/agribot/tower/.bloom/settings.json

For some reason bloom reads the target signature 0x552400 instead of 0x1e9221 which is of course wrong. I have no problem reading the target signature with AVRDude (it reads the correct sig 0x1e9221) . I have no problems using AVRDude for flashing, reading flash etc... In other words, No problems with the board.

I have erased flash before running bloom (and AVRDude) just in case to rule out that my firmware was messing with the RESET/UPDI pin (I have double checked it was erased by reading it back with AVRdude). Still same problem.

Here is my bloom.yaml

    debugLogging: true
    environments:
      default:
        shutdownPostDebugSession: true
    
        tool:
          name: "xplained-nano"
    
        target:
          name: "attiny416"
          physicalInterface: "updi"
          hardwareBreakpoints: true
    
        server:
          name: "avr-gdb-rsp"
          ipAddress: "127.0.0.1"
          port: 1442
          rangeStepping: true
    
    insight:
      activateOnStartup: false
      shutdownOnClose: true

Here is the output of AVRdude (read signature)

    avrdude -p attiny416 -c xplainedmini_updi -P usb -U signature:r:-:h
    
    avrdude: AVR device initialized and ready to accept instructions
    avrdude: device signature = 0x1e9221 (probably t416)
    
    avrdude: processing -U signature:r:-:h
    avrdude: reading signature memory ...
    avrdude: writing output file <stdout>
    0x1e,0x92,0x21
    
    avrdude done.  Thank you.

navnavnav, 2024-04-16:
Excellent, thanks for confirming. I will close this ticket once I've added the same change to the master branch.

navnavnav, 2025-02-23:
This has been fixed in v2.0.0.

closed 2025-02-23 by navnavnav