bloombloombloom/Bloom #111: Insight shows no TCA0 registers for the AVR128DA32

GeorgPerthil, 2024-07-23:
I have Bloom 1.0.0 installed on an Ubuntu 22.04.4 LTS. I can connect to the AVR (using mEDBG), but as mentioned in the title, there is no TCA0 shown in the target registers. I also tried an ATtiny1626, there is no TCA0 too. Maybe, it is because the TCA0 can be run in single and split mode? MPLAB-X shows both modes.

navnavnav, 2024-07-23:
Hi @GeorgPerthil,

Thanks for reporting this.

Ah yes, this appears to be due to the fact that the TCA module contains nested register groups, which Bloom doesn't support ATM. Those register groups are being ignored, and so the peripheral is considered to have no registers, which is why it's not appearing in the Insight GUI.

This issue should be fixed in the next release, as I have since refactored Bloom's TDF parser to accommodate nested register groups. Both the SINGLE and SPLIT registers from the TCA module will appear in the Insight GUI, as nested groups.

I still have a lot of work to do before the next release is ready, so I'm afraid you'll have to go without this fix for some time.

In the meantime, you could use GDB to inspect the register. Something like:

```
x/1bfx 0x08000A00
```

That should print the contents of the `CTRLA` register from the TCA0 peripheral, which is located at `0x00000A00`. Notice the OR'ing of the address with `0x08000000` - this is required by GDB for all peripheral register addresses.

Here's a list of the other registers for that peripheral:

|Absolute key|Name|Address space key|Address|Size (bytes)|Description|
|--- |--- |--- |--- |--- |--- |
|tca0.single.ctrla|CTRLA|data|0x00000A00|1|Control A|
|tca0.split.ctrla|CTRLA|data|0x00000A00|1|Control A|
|tca0.single.ctrlb|CTRLB|data|0x00000A01|1|Control B|
|tca0.split.ctrlb|CTRLB|data|0x00000A01|1|Control B|
|tca0.single.ctrlc|CTRLC|data|0x00000A02|1|Control C|
|tca0.split.ctrlc|CTRLC|data|0x00000A02|1|Control C|
|tca0.single.ctrld|CTRLD|data|0x00000A03|1|Control D|
|tca0.split.ctrld|CTRLD|data|0x00000A03|1|Control D|
|tca0.single.ctrleclr|CTRLECLR|data|0x00000A04|1|Control E Clear|
|tca0.split.ctrleclr|CTRLECLR|data|0x00000A04|1|Control E Clear|
|tca0.single.ctrleset|CTRLESET|data|0x00000A05|1|Control E Set|
|tca0.split.ctrleset|CTRLESET|data|0x00000A05|1|Control E Set|
|tca0.single.ctrlfclr|CTRLFCLR|data|0x00000A06|1|Control F Clear|
|tca0.single.ctrlfset|CTRLFSET|data|0x00000A07|1|Control F Set|
|tca0.single.evctrl|EVCTRL|data|0x00000A09|1|Event Control|
|tca0.single.intctrl|INTCTRL|data|0x00000A0A|1|Interrupt Control|
|tca0.split.intctrl|INTCTRL|data|0x00000A0A|1|Interrupt Control|
|tca0.single.intflags|INTFLAGS|data|0x00000A0B|1|Interrupt Flags|
|tca0.split.intflags|INTFLAGS|data|0x00000A0B|1|Interrupt Flags|
|tca0.single.dbgctrl|DBGCTRL|data|0x00000A0E|1|Degbug Control|
|tca0.split.dbgctrl|DBGCTRL|data|0x00000A0E|1|Degbug Control|
|tca0.single.temp|TEMP|data|0x00000A0F|1|Temporary data for 16-bit Access|
|tca0.single.cnt|CNT|data|0x00000A20|2|Count|
|tca0.split.lcnt|LCNT|data|0x00000A20|1|Low Count|
|tca0.split.hcnt|HCNT|data|0x00000A21|1|High Count|
|tca0.single.per|PER|data|0x00000A26|2|Period|
|tca0.split.lper|LPER|data|0x00000A26|1|Low Period|
|tca0.split.hper|HPER|data|0x00000A27|1|High Period|
|tca0.single.cmp0|CMP0|data|0x00000A28|2|Compare 0|
|tca0.split.lcmp0|LCMP0|data|0x00000A28|1|Low Compare|
|tca0.split.hcmp0|HCMP0|data|0x00000A29|1|High Compare|
|tca0.single.cmp1|CMP1|data|0x00000A2A|2|Compare 1|
|tca0.split.lcmp1|LCMP1|data|0x00000A2A|1|Low Compare|
|tca0.split.hcmp1|HCMP1|data|0x00000A2B|1|High Compare|
|tca0.single.cmp2|CMP2|data|0x00000A2C|2|Compare 2|
|tca0.split.lcmp2|LCMP2|data|0x00000A2C|1|Low Compare|
|tca0.split.hcmp2|HCMP2|data|0x00000A2D|1|High Compare|
|tca0.single.perbuf|PERBUF|data|0x00000A36|2|Period Buffer|
|tca0.single.cmp0buf|CMP0BUF|data|0x00000A38|2|Compare 0 Buffer|
|tca0.single.cmp1buf|CMP1BUF|data|0x00000A3A|2|Compare 1 Buffer|
|tca0.single.cmp2buf|CMP2BUF|data|0x00000A3C|2|Compare 2 Buffer|

navnavnav, 2024-07-23:
To inspect all registers in that peripheral:

```
x/61bfx 0x00800A00
```

That will print all 61 bytes from the `0x00000A00` peripheral base address.

navnavnav, 2025-02-23:
This is fixed in the v2.0.0 release.

closed 2025-02-23 by navnavnav