bloombloombloom/Bloom #114: Invalid GDB RSP client

ashlynblack, 2024-08-11:
Not sure if this should be addressed by Bloom or Microchip, but bloom and avr-gdb from Microchip's latest avr8 toolchain appear to be incompatible.

[ERROR] Invalid GDB RSP client - GDB client does not support HW or SW breakpoints - closing connection

[bloom.log](https://github.com/user-attachments/files/16571962/bloom.log)

OS: Ubuntu 20.04.6 LTS x86_64
Bloom v1.0.0
GNU gdb (GDB) 8.1.0.20180409-git installed from package manager repository works.
GNU gdb (AVR_8_bit_GNU_Toolchain_3.7.0_1796) 7.8 installed from Microchip's website does not.

navnavnav, 2024-08-11:
Hi @ashlynblack 

Thanks for reporting this.

GDB 7.8 was released in 2014. The latest version is 15.1, released last month. It seems Microchip are shipping a severely  outdated version. Bloom is incompatible with any version of GDB predating 8.1.

IIRC, the gdb-avr Debian package was abandoned for a period, but it now has an active maintainer and they're shipping 13.4 in Debian 13 and Ubuntu 24.04 LTS.

I'm currently using GDB 12.1, which works great with AVR targets.

As for the issue with 7.8, this isn't something I'm willing to fix, I'm afraid. Maintaining support for a 10 year old version of GDB will be too much of a burden, so I'm not willing to do it.

It might be worth asking Microchip to update their distributed binary, but given that they've neglected to do it for over 10 years, I doubt that will change. I think their XC8 compiler is a branch of an old version of GCC, which may be the reason behind the decision to stick with the older tools from the toolchain. But I could be wrong about that.

ashlynblack, 2024-08-12:
Ah that's perfectly understandable, shame on Microchip!
I'll be changing to Debian 12 which has gdb-avr 12.1 soon so all good. :)