bloombloombloom/Bloom #118: Website down

leo60228, 2024-08-15:
https://bloom.oscillate.io seems to be down. I suppose this isn't an issue with the project itself, but I wasn't sure where else to report it.

navnavnav, 2024-08-15:
Hi @leo60228 

I'm sorry about this. The website works fine for me, so I suspect you've been caught up in a blanket ban that had to resort to, lately.

Could you please share your ISP details, or just your current IP address. I will adjust the firewall rules accordingly.

leo60228, 2024-08-15:
Ah, this does seem to be ISP-specific. I'm having issues on AT&T Fiber
(AS7018 AT&T Services, Inc).

navnavnav, 2024-08-15:
Hmm, that ISP has not been banned. It's possible that your specific IP address was banned, but I cannot confirm this without your IP.

leo60228, 2024-08-15:
I have DNS for my network set up at leoserv.l3.pm, if that helps?

navnavnav, 2024-08-16:
Hmm, it seems your IP address was indeed banned. 

I found your IP quite high up in my firewall rules, which means it was banned some time ago. And because of my log retention period, I no longer have the server logs that would shed light on the incident that triggered the ban.

So I don't know why your specific IP was banned.

I have removed the firewall rule, so access should be restored. Could you please confirm this?

leo60228, 2024-08-16:
It appears to be fixed. Thank you!

closed 2024-08-16 by leo60228