bloombloombloom/Bloom #120: Switching a PICkit4 or SNAP to AVR mode

MCUdude, 2024-08-28:
In the docs it's recommended to install MPLAB X v6.05 in order to switch a PICKit4 or SNAP from PIC mode to AVR mode. I know this is a hopeless workaround, so that's why we built this feature into Avrdude! It's as simple as [sending a few bytes using libhidapi](https://github.com/avrdudes/avrdude/blob/4fa07e563fbe7ce0fa3c467eff20fc9e04de5298/src/jtag3.c#L1693-L1734), so I'm sure this is something that could be incorporated into Bloom.

navnavnav, 2024-08-28:
Hi @MCUdude,

Thanks for this. It's something I've been meaning to do and will get to it eventually.

I will keep this ticket updated.

navnavnav, 2024-12-30:
Development work for this is complete. Bloom will now automatically switch from PIC mode to EDBG mode, when necessary.

Changes will be included in v2.0.0. Will close this ticket upon release.

I also plan to update the article on the website, as the info is outdated and could do with a refresh.

MCUdude, 2024-12-30:
Awesome! It's great to see other tools implementing this as well. Bear in mind that it's not always possible to switch to AVR mode if in bootloader. That's because the firmware _may_ be broken/corrupt, and that's why it's in bootloader mode.

BTW @MX682X has managed utilize the PICkit5 (And PICkit4/SNAP in PIC mode) with GDB for debugging purposes. Maybe this is something that could be interesting for the Bloom project as well?

https://github.com/MX682X/PK5-UPDI-GDB-Server

navnavnav, 2024-12-30:
> Bear in mind that it's not always possible to switch to AVR mode if in bootloader. That's because the firmware may be broken/corrupt, and that's why it's in bootloader mode.

Thanks for this. Yeah I think that's OK - Bloom will attempt to exit the bootloader mode, as AVRDUDE does, but failing that, Bloom will give up and report the error to the user. I will also mention something in the article RE the factory firmware not supporting AVR mode, and thus requiring an update.

> Maybe this is something that could be interesting for the Bloom project as well?

That's interesting, thanks for sharing. I have considered supporting the newer Microchip debug tools, but have decided to hold off until Microchip release the documentation for the underlying protocol. I understand the new tools operate by executing target-specific scripts, but I know very little about the underlying protocol used to transfer these scripts to the tool. And I'm not really willing to rely on reverse-engineering here.

If Microchip ever get around to releasing the documentation, like Atmel did for the EDBG protocol, then I would certainly consider spending the time to implement it.

navnavnav, 2024-12-30:
@MCUdude could you please confirm something for me: Which version of AVRDUDE introduced the PIC/AVR mode switching functionality? Looking at the release logs, it seems to be version 7.3, but just want to be sure.

In the article, I plan to replace the MPLAB IPE instructions with AVRDUDE instructions, so I just want to make clear the minimum version of AVRDUDE that is required to perform the mode switch.

Thanks!

MX682X, 2024-12-30:
> If Microchip ever get around to releasing the documentation, like Atmel did for the EDBG protocol, then I would certainly consider spending the time to implement it.

In case you decide to implement it even without the official documentation - I did the reverse engeneering already, I'll be glad to help.

The code I've written to support PICkit5 in avrdude is basically a port of what I've written for the GDB Server in Python to C, so using that could be easier. Technically, there isn't much to add to support debugging, as it uses the same framework.

MCUdude, 2024-12-30:
> @MCUdude could you please confirm something for me: Which version of AVRDUDE introduced the PIC/AVR mode switching functionality? Looking at the release logs, it seems to be version 7.3, but just want to be sure.

You're correct, it was implemented in Avrdude 7.3. So both 7.3 and 8.0 is suitable. However, there is no reason not to use Avrdude 8.0.

navnavnav, 2024-12-30:
> So both 7.3 and 8.0 is suitable
Excellent, thanks!

navnavnav, 2025-02-23:
These changes have been included in the v2.0.0 release.

closed 2025-02-23 by navnavnav