bloombloombloom/Bloom #121: Waveshare Atmel ICE-C: "Activate physical interface command failed - Failure reason: Illegal target ID"

toby1984, 2024-11-17:
Hi,

I'm trying to get bloom working with an Atmega328p and the Waveshare Atmel ICE-C interface (which contains an original Atmel ICE PCB, https://www.waveshare.com/atmel-ice-c.htm).

Operating system is Ubuntu 24.04.01 LTS on x64

Configuration:
`
environments:
  default:
    tool:
      name: "atmel-ice"

    target:
      name: "atmega328p"
      physicalInterface: "debug-wire"
      # disableDebugWirePreDisconnect: true
      manageDwenFuseBit: true
      # cycleTargetPowerPostDwenUpdate: true
server:
  name: "avr-gdb-rsp"
  ipAddress: "127.0.0.1"
  port: 1442

insight:
  activateOnStartup: true`

Setting the DWEN fuse seems to have worked correctly and device signature is correct for an Atmega328p.

`$ bloom
2024-11-17 11:35:07.814 CET [MT]: [INFO] Selected environment: "default"
2024-11-17 11:35:07.814 CET [TC]: [INFO] Starting TargetController
2024-11-17 11:35:07.820 CET [TC]: [INFO] Connecting to debug tool
2024-11-17 11:35:07.971 CET [TC]: [INFO] Debug tool connected
2024-11-17 11:35:07.971 CET [TC]: [INFO] Debug tool name: Atmel-ICE
2024-11-17 11:35:08.008 CET [TC]: [INFO] Debug tool serial: J42700068591
2024-11-17 11:35:08.012 CET [TC]: [INFO] Activating target
2024-11-17 11:35:11.981 CET [TC]: [WARNING] Failed to activate the debugWire physical interface - attempting to access target via the ISP interface, for DWEN fuse bit inspection.
2024-11-17 11:35:11.981 CET [TC]: [INFO] Initiating ISP interface
2024-11-17 11:35:12.068 CET [TC]: [WARNING] Updating the DWEN fuse bit is a potentially dangerous operation. Bloom is provided "AS IS", without warranty of any kind. You are using Bloom at your own risk. In no event shall the copyright owner or contributors be liable for any damage caused as a result of using Bloom. For more details, see the Bloom license at https://bloom.oscillate.io/license
2024-11-17 11:35:12.068 CET [TC]: [INFO] Reading target signature via ISP
2024-11-17 11:35:12.177 CET [TC]: [INFO] Target signature confirmed: 0x1e950f
2024-11-17 11:35:12.215 CET [TC]: [INFO] Current SPIEN fuse bit value confirmed
2024-11-17 11:35:12.252 CET [TC]: [INFO] Cleared lock bits confirmed
2024-11-17 11:35:12.252 CET [TC]: [WARNING] Updating DWEN fuse bit
2024-11-17 11:35:12.325 CET [TC]: [INFO] DWEN fuse bit successfully updated
2024-11-17 11:35:12.362 CET [TC]: [INFO] Retrying debugWire physical interface activation
2024-11-17 11:35:12.700 CET [TC]: [ERROR] The TargetController encountered a fatal error. See below for errors:
2024-11-17 11:35:12.700 CET [TC]: [ERROR] AVR8 Activate physical interface command failed - Failure reason: Illegal target ID
2024-11-17 11:35:12.700 CET [TC]: [INFO] Shutting down TargetController
2024-11-17 11:35:12.700 CET [TC]: [INFO] Deactivating target
2024-11-17 11:35:12.776 CET [TC]: [INFO] Closing debug tool
2024-11-17 11:35:12.814 CET [MT]: [ERROR] TargetController failed to start up
2024-11-17 11:35:12.814 CET [MT]: [INFO] Shutting down Bloom
2024-11-17 11:35:12.814 CET [SH]: [INFO] Shutting down SignalHandler
`

navnavnav, 2024-11-17:
Hi @toby1984,

My first guess would be that the debug tool is having trouble communicating to the target via the debugWIRE pin.

A few questions:

- What clock frequency is your target running at? Have you tried lowering it?
- What else do you have connected to the debugWIRE pin (PC6)? Have you tried isolating that pin to rule out any interfering components in your circuit?
- From where did you acquire this target? Is it a genuine part?
- Do you have similar issues with other software such as AVRDUDE? Can you try cycling target power and then running `avrdude -c atmelice_dw -p m328p`. Is AVRDUDE able to read the correct signature then? If that works, try running Bloom again

toby1984, 2024-11-17:
Hey @navnavnav,

Thank you very much for the quick reply.
* Target is running at 16 MHz
* Nothing else connected to the RST PIN (AFAIK)
* Target is an non-genuine Arduino Nano (https://www.az-delivery.de/products/nano-v3-mit-ch340-arduino-kompatibel) but I've been using them for years with avrdude and a Diamex AVR ISP,  never had any issues
* trying to use `avrdude` with `-c atmelice_dw` unfortunately does not work either (DWEN fuse seems to be set though as my other ISP programmer can no longer talk to the device)

```
tobi@tobihome:~/arduino_workspace/transmitter(master)$ avrdude -v -c atmelice_dw -p m328p 

avrdude: Version 7.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /etc/avrdude.conf
         User configuration file is /home/tobi/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : atmelice_dw
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_DW
         Description     : Atmel-ICE (ARM/AVR) in debugWIRE mode
         ICE HW version  : 0
         ICE FW version  : 1.32 (rel. 117)
         Serial number   : J42700068591
         Vtarget         : 4.76 V

avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude main() error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower ISP clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude done.  Thank you.
```
Maybe I should look for/try flashing a newer firmware on the Atmel ICE ? I bought it only a couple of days ago and did not try hooking it up to Microship Studio on Windows (which as I understand comes with some firmware utility).

toby1984, 2024-11-17:
Sorry for the noise ... I just learned that on Arduino Nano (and clones) there DTR line of the USB-to-serial chip is connected to RST via a 10nF capacitor....

closed 2024-11-17 by toby1984

navnavnav, 2024-11-17:
> Maybe I should look for/try flashing a newer firmware on the Atmel ICE ? I bought it only a couple of days ago and did not try hooking it up to Microship Studio on Windows (which as I understand comes with some firmware utility).

Yeah that would be a good idea. While you're at it, you should also try debugging in Microchip Studio, to see if it works there.

Hmm, looking at the product page you provided, someone mentioned the need to remove a capacitor on PC6, in order to allow for debugWIRE communication, but doing so will break USB communication:

[image]
(that's the translated version)

Have you tried that?

Another review claims that the board possess a non-original chip. I'm not sure if they're referring to the ATmega328P, or some other chip, but worth noting.

[image]

If the ATmega328P is non-genuine, then that could explain the issue you're having, where communication via the ISP interface works fine, but not via the debugWIRE interface. The debugWIRE spec is proprietary, so I doubt any non-genuine parts have an implementation for the debugWIRE interface.

Either way, I still think you should try the firmware update/debug attempt in Microchip Studio. Let me know how it goes :)

navnavnav, 2024-11-17:
> Sorry for the noise ... I just learned that on Arduino Nano (and clones) the DTR line of the USB-to-serial chip is connected to RST via a 10nF capacitor....

Ah I see, no problem :)

toby1984, 2024-11-17:
Desoldering the capacitor fixed it, works like a treat ! Thank you very much for your time and this great tool !