bloombloombloom/Bloom #126: More flash wear than necessary

felias-fogg, 2025-05-03:
Bloom imposes more flash wear than necessary when dealing with software breakpoints. The problem appears to be that Bloom sends "breakpoint clear" commands as soon as it gets them from GDB. 

GDB deletes all breakpoints after execution stops and reasserts them before starting execution again. The EDBG protocol document states that the breakpoint set and clear commands take effect only when the "next flow control command is executed."  Because the clear and set commands cancel out each other, once the user has set a software breakpoint, it should stay in flash memory. Unfortunately, GDB tries to overstep a breakpoint at the current position by first reasserting all breakpoints except for the one at the current position. Then it requests a single step. After that, it reasserts the breakpoint and then issues a "continue" command. Since a "step" command is also a "flow control command", the software breakpoint is removed before the "step" command and reasserted afterwards. In other words, for each stop at a breakpoint, two flash programming operations are executed. 

Since this all sounds theoretical, it would be good to see whether this happens in practice. As it turns out, it does. I traced the debugWIRE line of an ATmega328, and, in fact, I could see this behavior. 

Interestingly, the hardware debuggers are perfectly capable of overstepping a breakpoint. If you issue a "step" or "run" command at a location with a software breakpoint, the debugger executes the instruction offline in its OCD and then continues. Well, this holds only if it is a one-word instruction. In my debugWIRE gdbserver, I make use of it and additionally simulate all two-word instructions. This means zero flash wear when the program stops at a breakpoint. 

All that is true at least in the debugWIRE context. However, I would be surprised if it is different in the other debugging protocols.

I hope that this observation helps to improve Bloom.

navnavnav, 2025-05-03:
Hi @felias-fogg 

Thanks for reporting this.

This is something I've been aware of for some time.

GDB can be configured to prevent this behavior, using the `breakpoint always-inserted` setting. Enabling it via `set breakpoint always-inserted on` will prevent GDB from deleting all set breakpoints whenever the target comes to a halt. But it won't prevent the deleting of breakpoints at the current PC, when stepping, so it doesn't completely mitigate the issue.

Not long ago, the [Limitations](https://bloom.oscillate.io//docs/limitations) page recommended enabling the `breakpoint always-inserted` setting, but I removed that recommendation when I discovered other, weird behavior from GDB, when that setting was enabled. Unfortunately I cannot remember exactly what went wrong, I only remember that GDB would misbehave in certain scenarios, when that setting was enabled.

I'm not sure if this is something that should be addressed in Bloom. After all, Bloom's job is to service commands from GDB, and that's exactly what it's doing. GDB is setting/removing breakpoints excessively, so I would say this should be addressed in GDB.

With all of that said, I've already had to work around the delayed set/remove breakpoint behavior in Bloom's EDBG driver, so it may not require significant dev time to do address this issue.

I'll give this some more thought at a later date.

felias-fogg, 2025-05-03:
Hi navnavnav, 

I disagree entirely with you about whether this should be handled on the GDBserver level or by GDB itself.

First, a GDBserver's job is not translating GDB RSP commands one-to-one into EDBG/CMSIS-DAP commands. This does not work for the vFlash or X commands, and it does not work for breakpoint handling.

Second, GDB assumes that it deals with hardware breakpoints (Z1), but you (and I) translate that to software breakpoints (Z0). So, I guess, you would not find much sympathy if you declared to the GDB community that they should change things you misinterpreted. 

Third, I believe that minimizing flash wear is an essential aspect of embedded debugging, independently of the two points above.  So, a little bit of thought and effort is called for.

Fourth, it is really not that complicated. AVaRICE simply does not update the (lazy) breakpoint list before a single-step operation. I believe there are corner cases where this hurts. I simply protect the breakpoint when there is a single-step and update it when there is a subsequent 'continue' command (see repo on [dw-gdbserver](https://github.com/felias-fogg/dw-gdbserver)).

navnavnav, 2025-05-03:
> First, a GDBserver's job is not translating GDB RSP commands one-to-one into EDBG/CMSIS-DAP commands.

This is not what I said. I said Bloom's job is to service commands. GDB issues commands, Bloom services them. The issue here is that GDB is issuing a command to remove a breakpoint, when there is no need for the removal. GDB may lack the awareness on this, but that is a limitation of GDB, not Bloom. It's not Bloom's responsibility to determine whether GDB actually wants to do the thing that it just told Bloom to do.

> This does not work for the vFlash or X commands

Can you elaborate? Bloom services vFlash commands as described in the GDB RSP specification. The only exception to this is when Bloom may ignore the flash erase command in order to accommodate delta programming - something that is specific to Bloom and not accommodated by GDB, which is why I make an exception for it. Bloom doesn't support `X` commands.

> Second, GDB assumes that it deals with hardware breakpoints (Z1), but you (and I) translate that to software breakpoints (Z0).

This is actually due to another limitation in GDB. GDB assumes all flash memory is read-only and cannot support software breakpoints. If you disable GDB's `auto-hw` mechanism (meaning GDB will not fall back to HW breakpoints, for "read-only" addresses), GDB will present an error when attempting to set breakpoints, and it will make no attempt to actually set the breakpoint (no `Z` packet is sent to the server):

```
(gdb) set breakpoint auto-hw off
(gdb) b *0x192
(gdb) c
Continuing.
Warning:
Cannot insert breakpoint 0.
Cannot set software breakpoint at read-only address 0x192
Command aborted.
```

So what you've described here is just another instance where I've had to work around a limitation in GDB. This does not mean that I should now do the same for all other limitations in GDB, does it?

> So, I guess, you would not find much sympathy if you declared to the GDB community that they should change things you misinterpreted.

Except I didn't misinterpret anything in this case. Bloom is forced to ignore the breakpoint type in the `Z` packet. Furthermore, the breakpoint type doesn't affect GDB's behavior when resuming execution - it will remove the breakpoint at PC regardless of the type of breakpoint. So no, the GDB community cannot claim that the issue is due to misinterpretation of the breakpoint command.

> Third, I believe that minimizing flash wear is an essential aspect of embedded debugging

Which is why GDB should provide an additional breakpoint setting, allowing for the prevention of the unnecessary breakpoint removal. Like they do for the `always-on` setting.

> Fourth, it is really not that complicated.

That's a bold statement. Have you studied Bloom's codebase? Thoroughly? To the point where you're confident enough to tell the author what's complicated? 

There are considerations - some of which that are specific to Bloom - that I need to make before I can determine the time and effort required to implement a work-around for this issue. As the author, I am best placed to make this assessment, not you.

As I said in my previous comment, I will consider implementing a work-around for this issue, but I stand by my belief that this issue should be addressed in GDB, where the issue lies.

felias-fogg, 2025-05-04:
I guess we have two different philosophies here. When I write an interface between two software layers, I try to translate the intention of the client layer to the server layer, minimizing adverse effects (such as flash wear). So, the implementation and semantics are a given. 

My statement that "it is not really complicated" was not meant to suggest how much effort it will take to implement it in your code, but rather about its conceptual complexity. And this I consider as low.