bloombloombloom/Bloom #127: RPI Bookworm missing libprocps, can't start Bloom

lkoepsel, 2025-05-08:
I'm attempting to install Bloom on a Raspberry Pi using the Headless AARCH64.deb package. 

I've updated to the latest version of Bookworm on the Raspberry Pi 5. 

I installed using:
sudo apt install ./Bloom-headless-2.0.0-Linux-aarch64.deb

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'bloom' instead of './Bloom-headless-2.0.0-Linux-aarch64.deb'
The following NEW packages will be installed:
  bloom
0 upgraded, 1 newly installed, 0 to remove and 2 not upgraded.
Need to get 0 B/2,574 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 /home/lkoepsel/Bloom-headless-2.0.0-Linux-aarch64.deb bloom arm64 2.0.0 [2,574 kB]
Selecting previously unselected package bloom.
(Reading database ... 162197 files and directories currently installed.)
Preparing to unpack .../Bloom-headless-2.0.0-Linux-aarch64.deb ...
Unpacking bloom (2.0.0) ...
Setting up bloom (2.0.0) ...
N: Download is performed unsandboxed as root as file '/home/lkoepsel/Bloom-headless-2.0.0-Linux-aarch64.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
``

When I attempt to run ```bloom init`` or ```bloom --version```, i get the following error:
```
bloom: error while loading shared libraries: libprocps.so.8: cannot open shared object file: No such file or directory
```

apt is unable to find a libprocps-dev or libprocps. It can install libproc2-dev, however, installing this does not solve the issue.

navnavnav, 2025-05-08:
Hey Lief,

Thanks for reporting this.

Bloom uses an outdated version of the libprocps library, and it looks like that version has been dropped in the newer Debian release. The properly fix this, I'll need to move over to the newer major release for that library. This is something I've been meaning to do, just haven't found the time for it.

In the mean time, you could try to use these shared library objects to get around that error.

I extracted these shared object files from the older Bloom v1.0.0 aarch64 DEB package. If you extract the tar and put the files in `/opt/bloom/lib/`, that should resolve the issue for you. You'll need root privileges to put those files there.

[libprocps.tar.gz](https://github.com/user-attachments/files/20109080/libprocps.tar.gz)

Alternatively, you could build that old version of libprocps from source. That should also work.

navnavnav, 2025-05-08:
Hmm, hang on, that package should have included the libprocps shared libraries. I think something else could be wrong here.

Could you run the following commands and post the output:
 ```ls -al /usr/bin/bloom```
 ```ls -al /opt/bloom/bin```
 ```ldd /opt/bloom/bin/bloom``` 

That will help me understand what's going on here.

navnavnav, 2025-05-08:
Ok, problem solved.

I screwed up the aarch64 package by using an old packaging script, which wasn't invoking Bloom correctly.

I've just updated the package to correct this issue. Please uninstall the old package and install the new one which you can download via [Bloom's download page](https://bloom.oscillate.io/download).

Please let me know how that goes.

lkoepsel, 2025-05-08:
Works great!

Downloaded latest AARCH64 headless version, spent a bit of time changing my bloom.yaml file 😊 and everthing works!

Thanks for the quick response and I'll provide additional comments as to 2.0 later.

lkoepsel, 2025-06-08:
I believe this is closed. No issues with Bloom and Bookworm.

closed 2025-06-08 by lkoepsel