bloombloombloom/Bloom #128: Display of target registers using RPI Bookworm has issues

lkoepsel, 2025-06-08:
Hi,
I've finally started bring up a chip from scratch. I'm using a ATtiny13A and Bloom has been great. I did have to ensure I started with ISP then use debugWire, which is well-documented, I just didn't read. :)

When attempting to display variables, the display is in color and is lovely.

When attempting to print register values, the color commands are shown, instead of changing the colors. Here is a image file to show the issue:

[image]

Is there anything I can do to show the target registers properly? 

Thanks!
Lief

lkoepsel, 2025-06-08:
I did a remote connection to the RPi5 using Raspberry Pi's new connection wizard (which works, kinda).
Same results, here's a fuzzy jpeg:

[image]

navnavnav, 2025-06-08:
ok, thanks for that.

Just to clarify, you're SSHing into your Pi, and **then** running avr-gdb, right? So you're running the avr-gdb binary that's installed on your Pi, not the one that's installed on your Mac? Or, are you running the GDB that's installed on your Mac, and then having it connect to Bloom's GDB server that's running on your Pi?

The difference is important, because GDB's color commands are working fine, but the ones from Bloom are not. If the GDB you're running is built for Mac OS, then that could explain this (as that build of GDB is probably using the MacOS specific color codes). But if you're running the GDB client that's installed on your Pi - which is built for Linux, not Mac OS - then I'm at a loss as to why Bloom's color codes are not working, but the ones from GDB are...

I'm running the same version of GD (13.1) on my Pi4, but can't replicate the issue using the default terminal. Very odd.

lkoepsel, 2025-06-08:
The good news on the Manjaro laptop, is I can do Insight!  (It too has the color issues.)

Here's my setup
I ssh into my RPi5. So everything runs locally on the RPi.
1) One tab running ```bloom attiny13a```
2. Second tab running ```avr-gdb --tui```

navnavnav, 2025-06-08:
ahh, it's the --tui option! When I run GDB with that option, it escapes the color codes...

lkoepsel, 2025-06-08:
Confirmed on both Manjaro and RPi...

I like using tui, can you fix it? please? 🙂

I'm not in a hurry, as I can use the Manjaro version for a while...

navnavnav, 2025-06-08:
Hmm, I'm not sure this is something I can fix on my end.

When GDB is in TUI mode, it seems to be converting the color escape sequences to readable characters. A bug was reported some time ago, and a patch seems to have been accepted, but I suspect that patch only affected GDB's own output, not the output of the GDB server when handling `qRcmd` (monitor) commands.

See this bug report: https://sourceware.org/bugzilla/show_bug.cgi?id=14126#c6

I've posted a comment, informing them of the issue persisting when GDB outputs qRcmd responses.

In the meantime, I may make the color escape sequences optional via the GDB server configuration. At least then the output of the mon lr/rr/wr/wrb commands is more readable...

lkoepsel, 2025-06-08:
Re: TUI
I like what you've done with the mon lr/rr commands. I'll work with gdb the way it is w/o TUI. I believe I can go back and forth when needed.

Re: Download
I removed my headless version, remote connected (via Raspberry's connection) and attempted to run Bloom. I got the following error:
```bash
bloom --version
/opt/bloom/bin/bloom: error while loading shared libraries: libQt6UiTools.so.6: cannot open shared object file: No such file or directory
```
The above text was from a ssh window, however the message was the same as the GUI version. Before I attempted to install libQt... I thought it best to advise. This is a low priority now given I have another system work around.

navnavnav, 2025-06-08:
ah, looks like I've missed some dependencies in that package. My apologies.

This should sort it:

```
sudo apt-get install qt6-qpa-plugins libqt6gui6 libqt6widgets6 libqt6uitools6 libqt6svg6
```

That should install all the dependencies. After that, try running Bloom again. Let me know how that foes 👍🏽

navnavnav, 2025-06-08:
That looks fine. That "wayland" message is more of a warning than an error.

You should be able to start bloom and activate the insight GUI via GDB.

I'm assuming the device enumeration error is down to the tool not being connected?

lkoepsel, 2025-06-08:
Works great! Many thanks!

The speed is fine on a RPi5 with 8GB. The bigger issue is that the GUI connection is a bit fuzzy and it is slow. It's great for a backup and if i used a keyboard and monitor with the RPI5, i believe it would look fine. Add the dependencies and I believe you have a working Insight version for the Raspberry Pi, albeit one which requires the 5.

BTW - switching between TUI and not, is really easy...
Ctrl-X Ctrl-a => exit TUI
Ctrl-X a => enter TUI

I'm testing on a 13 year old Mac running Arch Linux (Manjaro) and its great.

For my classes, I'll have the students use the RPi headless. For my work, I'll use a combination of MacBook Pro 2012 and RPi headless.

Thanks again!

closed 2025-06-08 by navnavnav

navnavnav, 2025-06-13:
Just FYI @lkoepsel: Looks like the [GDB maintainers have reproduced the issue](https://sourceware.org/bugzilla/show_bug.cgi?id=14126#c10), so hopefully we'll get a fix in a newer version of GDB 👍🏽

navnavnav, 2025-07-01:
Hey @lkoepsel 

Some good news RE that issue with color escape sequences in GDB: a patch has been accepted and applied to the master branch. I just built GDB from that branch and can confirm that the ANSI colors are now working as expected in TUI mode:

[image]

Looks like this fix will be available in the 17.1 release. But not sure when it will be available via Debian package repositories. You could always build it from source, if you need it sooner 👍🏽

lkoepsel, 2025-07-02:
That is great news! I’ve been building GDB from scratch as Debian is at 13.3, which seems to have a problem with Bloom. I’m using 16.3. 

I’ll build using 17.1 when it’s released. 

Thanks for letting me know.